jocover/jetson-ffmpeg #1: decoder coredump

st7109, 2019-09-19:
Hello, Sir, 
thanks for your great work and kindly share, I test this with Nano, just get below info:

Input #0, h264, from '/home/work/workshop/tegra_multimedia_api/data/Video/sample_outdoor_car_1080p_10fps.h264':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: h264 (High), yuv420p(progressive), 1920x1080 [SAR 1:1 DAR 16:9], 10 fps, 10 tbr, 1200k tbn, 20 tbc
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 261 
NVMEDIA: Reading vendor.tegra.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 261 
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (h264_nvmpi) -> wrapped_avframe (native))
Press [q] to stop, [?] for help
Output #0, null, to 'pipe:':
  Metadata:
    encoder         : Lavf58.29.100
    Stream #0:0: Video: wrapped_avframe, yuv420p, 1920x1080 [SAR 1:1 DAR 16:9], q=2-31, 200 kb/s, 10 fps, 10 tbn, 10 tbc
    Metadata:
      encoder         : Lavc58.54.100 wrapped_avframe
frame=  568 fps=146 q=-0.0 Lsize=N/A time=00:00:58.80 bitrate=N/A speed=15.1x    
video:297kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: unknown
NVMAP_IOC_READ failed: Interrupted system call
NVMAP_IOC_READ: Offset 0 SrcStride 2048 pDst 0x7fa5180130 DstStride 1920 Count 1080
nvbuf_utils: dmabuf_fd 1057 mapped entry NOT found
Segmentation fault (core dumped)

Did you hit this? my platform info is shown as below:
 Ubuntu 18.04.3 LTS
 Kernel Version: 4.9.140-tegra
 CUDA 10.0.326

jocover, 2019-09-19:
thank you for your feedback
I committed a patch https://github.com/jocover/jetson-ffmpeg/commit/9152a4e5eca17c2de19eb667f7fbb6165472168e

st7109, 2019-09-19:
I tested this again w/ patch, can duplicate same issue....error info is shown as below:

      encoder         : Lavc58.54.100 wrapped_avframe
frame=  568 fps=151 q=-0.0 Lsize=N/A time=00:00:58.80 bitrate=N/A speed=15.6x    
video:297kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: unknown
work@work-desktop:~/zxq/vsb_mw/ffmpeg$ ./ffmpeg -c:v h264_nvmpi -i ../../../workshop/tegra_multimedia_api/data/Video/sample_outdoor_car_1080p_10fps.h264 -f null -
ffmpeg version n4.2.1 Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 7 (Ubuntu/Linaro 7.4.0-1ubuntu1~18.04.1)
  configuration: --prefix=/home/work/zxq/vsb_mw/workspace --extra-cflags=-I/home/work/zxq/vsb_mw/workspace/include --extra-ldflags=-L/home/work/zxq/vsb_mw/workspace/lib --extra-libs='-lpthread -lm' --disable-shared --disable-debug --enable-static --disable-ffplay --disable-doc --enable-nvmpi --enable-pthreads
  libavutil      56. 31.100 / 56. 31.100
  libavcodec     58. 54.100 / 58. 54.100
  libavformat    58. 29.100 / 58. 29.100
  libavdevice    58.  8.100 / 58.  8.100
  libavfilter     7. 57.100 /  7. 57.100
  libswscale      5.  5.100 /  5.  5.100
  libswresample   3.  5.100 /  3.  5.100
Input #0, h264, from '../../../workshop/tegra_multimedia_api/data/Video/sample_outdoor_car_1080p_10fps.h264':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: h264 (High), yuv420p(progressive), 1920x1080 [SAR 1:1 DAR 16:9], 10 fps, 10 tbr, 1200k tbn, 20 tbc
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 261 
NVMEDIA: Reading vendor.tegra.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 261 
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (h264_nvmpi) -> wrapped_avframe (native))
Press [q] to stop, [?] for help
Output #0, null, to 'pipe:':
  Metadata:
    encoder         : Lavf58.29.100
    Stream #0:0: Video: wrapped_avframe, yuv420p, 1920x1080 [SAR 1:1 DAR 16:9], q=2-31, 200 kb/s, 10 fps, 10 tbn, 10 tbc
    Metadata:
      encoder         : Lavc58.54.100 wrapped_avframe
frame=  568 fps=148 q=-0.0 Lsize=N/A time=00:00:58.80 bitrate=N/A speed=15.3x    
video:297kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: unknown
nvbuf_utils: dmabuf_fd 1057 mapped entry NOT found
nvbuf_utils: Can not get HW buffer from FD... Exiting...
nvbuf_utils: dmabuf_fd -1 mapped entry NOT found
nvbuf_utils: Can not get HW buffer from FD... Exiting...
NvBufferGetParams failed for dmabuf_fd
nvbuf_utils: dmabuf_fd -1 mapped entry NOT found
nvbuf_utils: Can not get HW buffer from FD... Exiting...
NvBufferGetParams failed for dmabuf_fd
nvbuf_utils: dmabuf_fd -1 mapped entry NOT found
nvbuf_utils: Can not get HW buffer from FD... Exiting...
NvBufferGetParams failed for dmabuf_fd
nvbuf_utils: dmabuf_fd 1828717610 mapped entry NOT found
nvbuf_utils: Can not get HW buffer from FD... Exiting...

st7109, 2019-09-20:
another thing, this is only for nano, can't use with Jetson Xavier

closed 2019-09-20 by st7109