jocover/jetson-ffmpeg #2: ./ffmpeg doesn't run

SatyaNekkanti, 2019-09-20:
Hi Sir,

I was trying to compile ffmpeg in jetson-nano.
I was getting this error after compiling it and running ./ffmpeg
ffmpeg: error while loading shared libraries: libnvmpi.so.1.0.0: cannot open shared object file: No such file or directory

jocover, 2019-09-21:
Please use `ldconfig -p| grep nvmpi` find libnvmpi.so.1.0.0 path 
 /usr/local/lib directory/ is the default installation path

libnvmpi.so.1.0.0 (libc6,AArch64) => /usr/local/lib/libnvmpi.so.1.0.0
libnvmpi.so (libc6,AArch64) => /usr/local/lib/libnvmpi.so

SatyaNekkanti, 2019-09-21:
Hi Sir,
Thank you, I was able to run the ffmpeg.
Now I have trouble with decoder code. Core is getting dumped
I have tried to decode an RTSP link but I am getting this error

>ffmpeg version n4.2.1 Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 7 (Ubuntu/Linaro 7.4.0-1ubuntu1~18.04.1)
  configuration: --enable-nvmpi
  libavutil      56. 31.100 / 56. 31.100
  libavcodec     58. 54.100 / 58. 54.100
  libavformat    58. 29.100 / 58. 29.100
  libavdevice    58.  8.100 / 58.  8.100
  libavfilter     7. 57.100 /  7. 57.100
  libswscale      5.  5.100 /  5.  5.100
  libswresample   3.  5.100 /  3.  5.100
[rtsp @ 0x55a2d91770] max delay reached. need to consume packet
[rtsp @ 0x55a2d91770] RTP: missed 10 packets
[h264 @ 0x55a2d94ad0] Invalid level prefix
[h264 @ 0x55a2d94ad0] error while decoding MB 62 38
[h264 @ 0x55a2d94ad0] concealing 547 DC, 547 AC, 547 MV errors in I frame
Input #0, rtsp, from 'rtsp://--someRTSP link----':
  Metadata:
    title           : Media Presentation
  Duration: N/A, start: 0.540000, bitrate: N/A
    Stream #0:0: Video: h264 (Baseline), yuv420p(progressive), 1280x720, 10 fps, 25 tbr, 90k tbn, 20 tbc
Failed to query video capabilities: Inappropriate ioctl for device
Opening in BLOCKING MODE
NvMMLiteOpen : Block : BlockType = 261
NVMEDIA: Reading sys.display-size : status: 6
NvMMLiteBlockCreate : Block : BlockType = 261
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (h264_nvmpi) -> rawvideo (native))
Press [q] to stop, [?] for help
NVMAP_IOC_READ failed: Interrupted system call
NVMAP_IOC_READ: Offset 0 SrcStride 1280 pDst 0x7f7c006330 DstStride 1280 Count 360
NVMAP_IOC_READ failed: Interrupted system call
NVMAP_IOC_READ: Offset 1048576 SrcStride 768 pDst 0x7f7c006350 DstStride 640 Count 360
NVMAP_IOC_READ failed: Invalid argument
NVMAP_IOC_READ: Offset 1441792 SrcStride 768 pDst 0x7f7c006370 DstStride 640 Count 1280
NVMAP_IOC_READ failed: Interrupted system call
NVMAP_IOC_READ: Offset 0 SrcStride 1280 pDst 0x7f7c007e00 DstStride 1280 Count 360
NVMAP_IOC_READ failed: Interrupted system call
NVMAP_IOC_READ: Offset 1048576 SrcStride 768 pDst 0x7f7c007e20 DstStride 640 Count 360
NVMAP_IOC_READ failed: Invalid argument
NVMAP_IOC_READ: Offset 1441792 SrcStride 768 pDst 0x7f7c007e40 DstStride 640 Count 1280
Segmentation fault (core dumped)

**I have followed the installation steps, I even installed the patch. Still I get this error.**
Really appreciate if I this issue resolves.