jocover/jetson-ffmpeg #6: h264_nvmpi encoding fails with error

danieloneill, 2019-10-19:
Running on:

Linux version 4.9.140-tegra (buildbrain@mobile-u64-911) (gcc version 7.3.1 20180425 [linaro-7.3-2018.05 revision d29120a424ecfbc167ef90065c0eeb7f91977701] (Linaro GCC 7.3-2018.05) ) #1 SMP PREEMPT Tue Jul 16 17:04:49 PDT 2019

```
doneill@jetson:~$ ffmpeg -re -vcodec h264_nvmpi -i ./2019-01-06\ 04-07-25.mkv -r 30 -profile:v high -level 4.2 -b:v 720k -ar 44100 -ab 128k -threads 4 -vcodec h264_nvmpi -acodec libmp3lame -flags +global_header -f matroska test.mkv
ffmpeg version git-2019-10-16-1e35519 Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 7 (Ubuntu/Linaro 7.4.0-1ubuntu1~18.04.1)
  configuration: --enable-nvmpi --enable-cuda-nvcc --enable-nonfree --enable-shared --enable-gpl --nvcc=/usr/local/cuda-10.0/bin/nvcc --enable-libmp3lame --enable-libopus --enable-librtmp --enable-libx264
  libavutil      56. 35.101 / 56. 35.101
  libavcodec     58. 59.102 / 58. 59.102
  libavformat    58. 33.100 / 58. 33.100
  libavdevice    58.  9.100 / 58.  9.100
  libavfilter     7. 63.100 /  7. 63.100
  libswscale      5.  6.100 /  5.  6.100
  libswresample   3.  6.100 /  3.  6.100
  libpostproc    55.  6.100 / 55.  6.100
Input #0, matroska,webm, from './2019-01-06 04-07-25.mkv':
  Metadata:
    ENCODER         : Lavf57.83.100
  Duration: 00:05:24.53, start: 0.000000, bitrate: 3627 kb/s
    Stream #0:0: Video: h264 (Main), yuv420p(tv, bt470bg/unknown/unknown, progressive), 1280x720 [SAR 1:1 DAR 16:9], 30 fps, 30 tbr, 1k tbn, 60 tbc (default)
    Metadata:
      DURATION        : 00:05:24.533000000
    Stream #0:1: Audio: aac (LC), 48000 Hz, stereo, fltp (default)
    Metadata:
      title           : Track1
      DURATION        : 00:05:24.480000000
File 'test.mkv' already exists. Overwrite? [y/N] y
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 261 
NVMEDIA: Reading vendor.tegra.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 261 
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (h264_nvmpi) -> h264 (h264_nvmpi))
  Stream #0:1 -> #0:1 (aac (native) -> mp3 (libmp3lame))
Press [q] to stop, [?] for help
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 4 
===== NVMEDIA: NVENC =====
NvMMLiteBlockCreate : Block : BlockType = 4 
875967048
842091865
H264: Profile = 100, Level = 42 
Could not write header for output file #0 (incorrect codec parameters ?): Invalid data found when processing input
Error initializing output stream 0:0 -- 
Segmentation fault (core dumped)
doneill@jetson:~$ 
```

Using libx264 works as expected:

```
doneill@jetson:~$ ffmpeg -re -vcodec h264_nvmpi -i ./2019-01-06\ 04-07-25.mkv -r 30 -profile:v high -level 4.2 -b:v 720k -ar 44100 -ab 128k -threads 4 -vcodec libx264 -acodec libmp3lame -flags +global_header -f matroska test.mkv
ffmpeg version git-2019-10-16-1e35519 Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 7 (Ubuntu/Linaro 7.4.0-1ubuntu1~18.04.1)
  configuration: --enable-nvmpi --enable-cuda-nvcc --enable-nonfree --enable-shared --enable-gpl --nvcc=/usr/local/cuda-10.0/bin/nvcc --enable-libmp3lame --enable-libopus --enable-librtmp --enable-libx264
  libavutil      56. 35.101 / 56. 35.101
  libavcodec     58. 59.102 / 58. 59.102
  libavformat    58. 33.100 / 58. 33.100
  libavdevice    58.  9.100 / 58.  9.100
  libavfilter     7. 63.100 /  7. 63.100
  libswscale      5.  6.100 /  5.  6.100
  libswresample   3.  6.100 /  3.  6.100
  libpostproc    55.  6.100 / 55.  6.100
Input #0, matroska,webm, from './2019-01-06 04-07-25.mkv':
  Metadata:
    ENCODER         : Lavf57.83.100
  Duration: 00:05:24.53, start: 0.000000, bitrate: 3627 kb/s
    Stream #0:0: Video: h264 (Main), yuv420p(tv, bt470bg/unknown/unknown, progressive), 1280x720 [SAR 1:1 DAR 16:9], 30 fps, 30 tbr, 1k tbn, 60 tbc (default)
    Metadata:
      DURATION        : 00:05:24.533000000
    Stream #0:1: Audio: aac (LC), 48000 Hz, stereo, fltp (default)
    Metadata:
      title           : Track1
      DURATION        : 00:05:24.480000000
File 'test.mkv' already exists. Overwrite? [y/N] y
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 261 
NVMEDIA: Reading vendor.tegra.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 261 
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (h264_nvmpi) -> h264 (libx264))
  Stream #0:1 -> #0:1 (aac (native) -> mp3 (libmp3lame))
Press [q] to stop, [?] for help
[libx264 @ 0x55a2168a30] using SAR=1/1
[libx264 @ 0x55a2168a30] using cpu capabilities: ARMv8 NEON
[libx264 @ 0x55a2168a30] profile High, level 4.2
[libx264 @ 0x55a2168a30] 264 - core 152 r2854 e9a5903 - H.264/MPEG-4 AVC codec - Copyleft 2003-2017 - http://www.videolan.org/x264.html - options: cabac=1 ref=3 deblock=1:0:0 analyse=0x3:0x113 me=hex subme=7 psy=1 psy_rd=1.00:0.00 mixed_ref=1 me_range=16 chroma_me=1 trellis=1 8x8dct=1 cqm=0 deadzone=21,11 fast_pskip=1 chroma_qp_offset=-2 threads=4 lookahead_threads=1 sliced_threads=0 nr=0 decimate=1 interlaced=0 bluray_compat=0 constrained_intra=0 bframes=3 b_pyramid=2 b_adapt=1 b_bias=0 direct=1 weightb=1 open_gop=0 weightp=2 keyint=250 keyint_min=25 scenecut=40 intra_refresh=0 rc_lookahead=40 rc=abr mbtree=1 bitrate=720 ratetol=1.0 qcomp=0.60 qpmin=0 qpmax=69 qpstep=4 ip_ratio=1.40 aq=1:1.00
Output #0, matroska, to 'test.mkv':
  Metadata:
    encoder         : Lavf58.33.100
    Stream #0:0: Video: h264 (libx264) (H264 / 0x34363248), yuv420p, 1280x720 [SAR 1:1 DAR 16:9], q=-1--1, 720 kb/s, 30 fps, 1k tbn, 30 tbc (default)
    Metadata:
      DURATION        : 00:05:24.533000000
      encoder         : Lavc58.59.102 libx264
    Side data:
      cpb: bitrate max/min/avg: 0/0/720000 buffer size: 0 vbv_delay: N/A
    Stream #0:1: Audio: mp3 (libmp3lame) (U[0][0][0] / 0x0055), 44100 Hz, stereo, fltp, 128 kb/s (default)
    Metadata:
      title           : Track1
      DURATION        : 00:05:24.480000000
      encoder         : Lavc58.59.102 libmp3lame
frame=  398 fps= 17 q=-1.0 Lsize=    1275kB time=00:00:13.42 bitrate= 777.8kbits/s speed=0.567x    
video:1057kB audio:210kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.573169%
[libx264 @ 0x55a2168a30] frame I:2     Avg QP:34.75  size: 20989
[libx264 @ 0x55a2168a30] frame P:113   Avg QP:39.36  size:  4858
[libx264 @ 0x55a2168a30] frame B:283   Avg QP:42.74  size:  1735
[libx264 @ 0x55a2168a30] consecutive B-frames:  0.8%  6.0% 21.9% 71.4%
[libx264 @ 0x55a2168a30] mb I  I16..4: 36.3% 53.0% 10.7%
[libx264 @ 0x55a2168a30] mb P  I16..4: 11.9% 25.0%  0.3%  P16..4: 20.7%  1.3%  0.6%  0.0%  0.0%    skip:40.1%
[libx264 @ 0x55a2168a30] mb B  I16..4:  1.1%  1.6%  0.0%  B16..8: 24.6%  0.7%  0.0%  direct: 0.8%  skip:71.2%  L0:46.0% L1:53.4% BI: 0.6%
[libx264 @ 0x55a2168a30] final ratefactor: 37.98
[libx264 @ 0x55a2168a30] 8x8 transform intra:65.5% inter:93.4%
[libx264 @ 0x55a2168a30] coded y,uvDC,uvAC intra: 7.2% 32.9% 4.2% inter: 2.0% 6.2% 0.0%
[libx264 @ 0x55a2168a30] i16 v,h,dc,p: 36% 42%  9% 13%
[libx264 @ 0x55a2168a30] i8 v,h,dc,ddl,ddr,vr,hd,vl,hu: 15% 11% 65%  1%  2%  1%  2%  1%  1%
[libx264 @ 0x55a2168a30] i4 v,h,dc,ddl,ddr,vr,hd,vl,hu: 21% 25% 28%  4%  6%  4%  5%  4%  3%
[libx264 @ 0x55a2168a30] i8c dc,h,v,p: 75% 14% 10%  1%
[libx264 @ 0x55a2168a30] Weighted P-Frames: Y:0.0% UV:0.0%
[libx264 @ 0x55a2168a30] ref P L0: 63.9% 11.6% 17.0%  7.5%
[libx264 @ 0x55a2168a30] ref B L0: 85.0% 11.9%  3.2%
[libx264 @ 0x55a2168a30] ref B L1: 95.6%  4.4%
[libx264 @ 0x55a2168a30] kb/s:652.52
doneill@jetson:~$ 
```

I've also tried decoding with libx264 instead, to feed the encoder, but get the same results.

danieloneill, 2019-10-20:
Okay, mpegts and flv (with SPS/PPS after IDR enabled) works perfectly (after I rebooted for some reason!), but matroska format still yields the same error.  I did hack the sourcecode ```param.insert_spspps_idr=1; // (avctx->flags & AV_CODEC_FLAG_GLOBAL_HEADER)?0:1;``` because ```-v:flags global_header``` wasn't working for me.

jocover, 2019-10-20:
matroska using AV_CODEC_FLAG_GLOBAL_HEADER flag
but nvidia l4t encoder not support get video sps\pps api,so you need to manually generate spspps to avctx->extradata

this is test version patch
https://gist.github.com/jocover/c993a532d5de37011d1191138c9a28b1#file-ffmpeg_nvmpi-patch-L409

    uint8_t sps_buf[] = { 0x00, 0x00, 0x00, 0x01, 0x67, 0x42, 0x40, 0x33, 0x96, 0x54, 0x03, 0xC0, 0x11, 0x3F, 0x2A };
    uint8_t pps_buf[] = { 0x00, 0x00, 0x00, 0x01, 0x68, 0xCE, 0x3C, 0x80 };
	
    if ( avctx->flags & AV_CODEC_FLAG_GLOBAL_HEADER && (avctx->codec->id == AV_CODEC_ID_H264) )
    {
    	avctx->extradata_size	= sizeof(sps_buf) + sizeof(pps_buf);
    	avctx->extradata	= av_mallocz( avctx->extradata_size + AV_INPUT_BUFFER_PADDING_SIZE );
    	memcpy( avctx->extradata, sps_buf, sizeof(sps_buf) );
    	memcpy( avctx->extradata + sizeof(sps_buf), pps_buf, sizeof(pps_buf) );
    	memset( avctx->extradata + sizeof(sps_buf) + sizeof(pps_buf), 0, AV_INPUT_BUFFER_PADDING_SIZE );
    }

jocover, 2019-10-23:
Fixed matroska support issues
https://github.com/jocover/jetson-ffmpeg/commit/b5af32467959ca66ed9591f4520609cc9eddd46b

closed 2019-10-29 by jocover