jocover/jetson-ffmpeg #8: nvmpi-patched ffmpeg encode appears to hang for nonstandard resolutions

jacobgorm, 2019-10-29:
$ ffmpeg -loglevel debug -y -f rawvideo  -s  480x272 -i /dev/zero -c:v h264_nvmpi foo.mp4
ffmpeg version 61853f7 Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 7 (Ubuntu/Linaro 7.4.0-1ubuntu1~18.04.1)
  configuration: --prefix=/usr --enable-nvmpi --disable-gpl
  libavutil      56. 31.100 / 56. 31.100
  libavcodec     58. 54.100 / 58. 54.100
  libavformat    58. 29.100 / 58. 29.100
  libavdevice    58.  8.100 / 58.  8.100
  libavfilter     7. 57.100 /  7. 57.100
  libswscale      5.  5.100 /  5.  5.100
  libswresample   3.  5.100 /  3.  5.100
Splitting the commandline.
Reading option '-loglevel' ... matched as option 'loglevel' (set logging level) with argument 'debug'.
Reading option '-y' ... matched as option 'y' (overwrite output files) with argument '1'.
Reading option '-f' ... matched as option 'f' (force format) with argument 'rawvideo'.
Reading option '-s' ... matched as option 's' (set frame size (WxH or abbreviation)) with argument '480x272'.
Reading option '-i' ... matched as input url with argument '/dev/zero'.
Reading option '-c:v' ... matched as option 'c' (codec name) with argument 'h264_nvmpi'.
Reading option 'foo.mp4' ... matched as output url.
Finished splitting the commandline.
Parsing a group of options: global .
Applying option loglevel (set logging level) with argument debug.
Applying option y (overwrite output files) with argument 1.
Successfully parsed a group of options.
Parsing a group of options: input url /dev/zero.
Applying option f (force format) with argument rawvideo.
Applying option s (set frame size (WxH or abbreviation)) with argument 480x272.
Successfully parsed a group of options.
Opening an input file: /dev/zero.
[rawvideo @ 0x5588804780] Opening '/dev/zero' for reading
[file @ 0x5588805190] Setting default whitelist 'file,crypto'
[rawvideo @ 0x5588804780] Before avformat_find_stream_info() pos: 0 bytes read:32768 seeks:0 nb_streams:1
[rawvideo @ 0x5588804780] All info found
[rawvideo @ 0x5588804780] After avformat_find_stream_info() pos: 195840 bytes read:195840 seeks:0 frames:1
Input #0, rawvideo, from '/dev/zero':
  Duration: N/A, start: 0.000000, bitrate: 39168 kb/s
    Stream #0:0, 1, 1/25: Video: rawvideo, 1 reference frame (I420 / 0x30323449), yuv420p, 480x272, 0/1, 39168 kb/s, 25 tbr, 25 tbn, 25 tbc
Successfully opened the file.
Parsing a group of options: output url foo.mp4.
Applying option c:v (codec name) with argument h264_nvmpi.
Successfully parsed a group of options.
Opening an output file: foo.mp4.
[file @ 0x55887eb5a0] Setting default whitelist 'file,crypto'
Successfully opened the file.
Stream mapping:
  Stream #0:0 -> #0:0 (rawvideo (native) -> h264 (h264_nvmpi))
Press [q] to stop, [?] for help
cur_dts is invalid st:0 (0) [init:0 i_done:0 finish:0] (this is harmless if it occurs once at the start per stream)
[rawvideo @ 0x5588806b40] PACKET SIZE: 195840, STRIDE: 720
detected 4 logical cores
[graph 0 input from stream 0:0 @ 0x55887e3ee0] Setting 'video_size' to value '480x272'
[graph 0 input from stream 0:0 @ 0x55887e3ee0] Setting 'pix_fmt' to value '0'
[graph 0 input from stream 0:0 @ 0x55887e3ee0] Setting 'time_base' to value '1/25'
[graph 0 input from stream 0:0 @ 0x55887e3ee0] Setting 'pixel_aspect' to value '0/1'
[graph 0 input from stream 0:0 @ 0x55887e3ee0] Setting 'sws_param' to value 'flags=2'
[graph 0 input from stream 0:0 @ 0x55887e3ee0] Setting 'frame_rate' to value '25/1'
[graph 0 input from stream 0:0 @ 0x55887e3ee0] w:480 h:272 pixfmt:yuv420p tb:1/25 fr:25/1 sar:0/1 sws_param:flags=2
[format @ 0x55888162c0] Setting 'pix_fmts' to value 'yuv420p'
[AVFilterGraph @ 0x55887f79f0] query_formats: 4 queried, 3 merged, 0 already done, 0 delayed
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 4 
===== NVMEDIA: NVENC =====
NvMMLiteBlockCreate : Block : BlockType = 4 
875967048
842091865
H264: Profile = 77, Level = 51 

(And then it just sits, there, a non-patched ffmpeg works fine.)

jocover, 2019-10-30:
thank you for your testing

patch
https://github.com/jocover/jetson-ffmpeg/commit/e892195797ada3599a737265150559fbac8d0151

jacobgorm, 2019-10-31:
Unfortunately it just segfaults now:

Here is the backtrace from gdb:

(gdb) bt
#0  __GI___pthread_mutex_lock (mutex=0x8080808080808080) at pthread_mutex_lock.c:65
#1  0x0000007fb7710760 in NvOsMutexLock () from /usr/lib/aarch64-linux-gnu/tegra/libnvos.so
#2  0x0000007fab0a88bc in NvMMQueueEnQ () from /usr/lib/aarch64-linux-gnu/tegra/libnvmm_utils.so
#3  0x0000007fb3c4c25c in ?? () from /usr/lib/aarch64-linux-gnu/tegra/libtegrav4l2.so
#4  0x0000007fb3c4f394 in ?? () from /usr/lib/aarch64-linux-gnu/tegra/libtegrav4l2.so
#5  0x0000007fb3c43960 in TegraV4L2_Ioctl () from /usr/lib/aarch64-linux-gnu/tegra/libtegrav4l2.so
#6  0x0000007fb7fafeb4 in plugin_ioctl () from /usr/lib/aarch64-linux-gnu/libv4l/plugins/nv/libv4l2_nvvideocodec.so
#7  0x0000007fb7a3c5c0 in v4l2_ioctl () from /usr/lib/aarch64-linux-gnu/libv4l2.so.0
#8  0x0000007fb7d43334 in NvV4l2ElementPlane::qBuffer(v4l2_buffer&, NvBuffer*) () from /usr/local/lib/libnvmpi.so.1.0.0
#9  0x0000007fb7d3aaac in nvmpi_encoder_put_frame () from /usr/local/lib/libnvmpi.so.1.0.0
#10 0x0000005555bcfab0 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

jacobgorm, 2019-12-03:
The patch provided here https://github.com/jocover/jetson-ffmpeg/commit/e892195797ada3599a737265150559fbac8d0151 seems to make things worse, I would suggesting reverting it from master until a working solution can be found.

f1l1b0x, 2020-01-14:
is it required to scale on cuda? or is the multimedia api capable of doing custom framesize scaling?

aspecialkey, 2020-08-27:
hi, does somebody know, how to scale on hardware?