jocover/jetson-ffmpeg #9: Segmentation fault in decoding

baltov, 2019-10-30:
Hi, 
decoding is not working
It can not decode sample Jetson Nano file
./ffmpeg -c:v h264_nvmpi -i /usr/src/nvidia/tegra_multimedia_api/data/Video/sample_outdoor_car_1080p_10fps.h264 -f null - 

it throws Segmentation fault (core dumped)

jocover, 2019-10-31:
work fine on my jetson nano

jocover@jetson-nano:~/ffmpeg$ ./ffmpeg -c:v h264_nvmpi -i /usr/src/nvidia/tegra_multimedia_api/data/Video/sample_outdoor_car_1080p_10fps.h264 -f null -
ffmpeg version 61853f7 Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 7 (Ubuntu/Linaro 7.4.0-1ubuntu1~18.04.1)
  configuration: --enable-nvmpi
  libavutil      56. 31.100 / 56. 31.100
  libavcodec     58. 54.100 / 58. 54.100
  libavformat    58. 29.100 / 58. 29.100
  libavdevice    58.  8.100 / 58.  8.100
  libavfilter     7. 57.100 /  7. 57.100
  libswscale      5.  5.100 /  5.  5.100
  libswresample   3.  5.100 /  3.  5.100
Input #0, h264, from '/usr/src/nvidia/tegra_multimedia_api/data/Video/sample_outdoor_car_1080p_10fps.h264':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: h264 (High), yuv420p(progressive), 1920x1080 [SAR 1:1 DAR 16:9], 10 fps, 10 tbr, 1200k tbn, 20 tbc
Opening in BLOCKING MODE
NvMMLiteOpen : Block : BlockType = 261
NVMEDIA: Reading vendor.tegra.display-size : status: 6
NvMMLiteBlockCreate : Block : BlockType = 261
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (h264_nvmpi) -> wrapped_avframe (native))
Press [q] to stop, [?] for help
Output #0, null, to 'pipe:':
  Metadata:
    encoder         : Lavf58.29.100
    Stream #0:0: Video: wrapped_avframe, yuv420p, 1920x1080 [SAR 1:1 DAR 16:9], q=2-31, 200 kb/s, 10 fps, 10 tbn, 10 tbc
    Metadata:
      encoder         : Lavc58.54.100 wrapped_avframe
frame=  568 fps=187 q=-0.0 Lsize=N/A time=00:00:58.80 bitrate=N/A speed=19.3x
video:297kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: unknown

You can try to recompile latest version nvmpi and ffmpeg

baltov, 2019-10-31:
Thank you for quick answer.
Yes I compiled it with latest ffmpeg 4.2 as it is described.

 ./ffmpeg -c:v h264_nvmpi -i /usr/src/nvidia/tegra_multimedia_api/data/Video/sample_outdoor_car_1080p_10fps.h264 -f null - 
ffmpeg version 61853f7 Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 7 (Ubuntu/Linaro 7.4.0-1ubuntu1~18.04.1)
  configuration: --enable-nvmpi
  libavutil      56. 31.100 / 56. 31.100
  libavcodec     58. 54.100 / 58. 54.100
  libavformat    58. 29.100 / 58. 29.100
  libavdevice    58.  8.100 / 58.  8.100
  libavfilter     7. 57.100 /  7. 57.100
  libswscale      5.  5.100 /  5.  5.100
  libswresample   3.  5.100 /  3.  5.100
Input #0, h264, from '/usr/src/nvidia/tegra_multimedia_api/data/Video/sample_outdoor_car_1080p_10fps.h264':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: h264 (High), yuv420p(progressive), 1920x1080 [SAR 1:1 DAR 16:9], 10 fps, 10 tbr, 1200k tbn, 20 tbc
Failed to query video capabilities: Inappropriate ioctl for device
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 261 
NVMEDIA: Reading sys.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 261 
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (h264_nvmpi) -> wrapped_avframe (native))
Press [q] to stop, [?] for help
NVMAP_IOC_READ failed: Interrupted system call
NVMAP_IOC_READ: Offset 0 SrcStride 2048 pDst 0x7fa0005dc0 DstStride 1920 Count 540
NVMAP_IOC_READ failed: Interrupted system call
NVMAP_IOC_READ: Offset 2228224 SrcStride 1024 pDst 0x7fa0005de0 DstStride 960 Count 540
NVMAP_IOC_READ failed: Invalid argument
NVMAP_IOC_READ: Offset 2883584 SrcStride 1024 pDst 0x7fa0005e00 DstStride 960 Count 2048
NVMAP_IOC_READ failed: Interrupted system call
NVMAP_IOC_READ: Offset 0 SrcStride 2048 pDst 0x7fa0007890 DstStride 1920 Count 540
NVMAP_IOC_READ failed: Interrupted system call
NVMAP_IOC_READ: Offset 2228224 SrcStride 1024 pDst 0x7fa00078b0 DstStride 960 Count 540
NVMAP_IOC_READ failed: Invalid argument
NVMAP_IOC_READ: Offset 2883584 SrcStride 1024 pDst 0x7fa00078d0 DstStride 960 Count 2048
Segmentation fault (core dumped)

if it can be helpful, that is the stack trace

Thread 1 "ffmpeg_g" received signal SIGSEGV, Segmentation fault.
__memcpy_generic () at ../sysdeps/aarch64/multiarch/../memcpy.S:188
188	../sysdeps/aarch64/multiarch/../memcpy.S: No such file or directory.
(gdb) bt
#0  __memcpy_generic () at ../sysdeps/aarch64/multiarch/../memcpy.S:188
#1  0x0000005556113908 in memcpy (__len=1920, __src=0x7fa0020cc0, __dest=<optimized out>) at /usr/include/aarch64-linux-gnu/bits/string_fortified.h:34
#2  image_copy_plane (height=1026, bytewidth=1920, src_linesize=<optimized out>, 
    src=0x7fa0020cc0 "\230\230\227\227\226\225\225\224\223\221\216\214\212\210\205\203\200~{zywvuutuutsrqqqqqpooomiiknlijginornikquwwnjiggffiquz|||||xttqaXVX^fonfelsz|{{{{{{|||||}}}}}}}\177\200\200\177wi][\\]^^_]^\\itz{|}~\177\177\177\177~}}}|zwwxyyyyyzzzzzzytorz}}}|{zxxxy{}~\200\200\200\200\200\177\177~|{zzz|}~\177\201\203\203\203\203\200"..., 
    dst_linesize=<optimized out>, dst=<optimized out>) at libavutil/imgutils.c:317
#3  image_copy (copy_plane=<optimized out>, height=<optimized out>, width=1920, pix_fmt=AV_PIX_FMT_YUV420P, src_linesizes=0x7fffffe228, src_data=0x7fffffe2c8, 
    dst_linesizes=0x7fffffe208, dst_data=0x5556d56c40) at libavutil/imgutils.c:380
#4  av_image_copy (dst_data=dst_data@entry=0x5556d56c40, dst_linesizes=dst_linesizes@entry=0x5556d56c80, src_data=src_data@entry=0x7fffffe2c8, 
    src_linesizes=src_linesizes@entry=0x7fffffe2b8, pix_fmt=AV_PIX_FMT_YUV420P, width=1920, height=1080) at libavutil/imgutils.c:399
#5  0x0000005555bdb9b8 in nvmpi_decode (avctx=0x5556d561d0, data=0x5556d56c40, got_frame=0x7fffffe3e0, avpkt=0x5556d39dc0) **at libavcodec/nvmpi_dec.c:113**
#6  0x00000055559fc0e8 in decode_simple_internal (frame=0x5556d56c40, avctx=0x5556d561d0) at libavcodec/decode.c:433
#7  decode_simple_receive_frame (frame=<optimized out>, avctx=<optimized out>) at libavcodec/decode.c:629
#8  decode_receive_frame_internal (avctx=avctx@entry=0x5556d561d0, frame=0x5556d56c40) at libavcodec/decode.c:647
#9  0x00000055559fcba0 in avcodec_send_packet (avctx=0x5556d561d0, avpkt=0x7fffffe4a0) at libavcodec/decode.c:705
#10 0x00000055556e0908 in decode (pkt=0x7fffffe4a0, got_frame=0x7fffffe608, frame=<optimized out>, avctx=0x5556d561d0) at fftools/ffmpeg.c:2240
#11 decode_video (ist=0x5556d62010, pkt=0x7fffffe618, got_output=0x7fffffe608, duration_pts=0x7fffffe610, eof=0, decode_failed=0x7fffffe60c) at fftools/ffmpeg.c:2382
#12 0x00000055556e1450 in process_input_packet (ist=0x5556d62010, pkt=0x7fffffe848, no_eof=0) at fftools/ffmpeg.c:2623
#13 0x00000055556e3960 in process_input (file_index=<optimized out>) at fftools/ffmpeg.c:4518
#14 transcode_step () at fftools/ffmpeg.c:4638
#15 transcode () at fftools/ffmpeg.c:4692
#16 0x00000055556c2cb8 in main (argc=<optimized out>, argv=<optimized out>) at fftools/ffmpeg.c:4894