jocover/jetson-ffmpeg #11: Allow NV12 choice

xsacha, 2019-11-02:
+	nvmpi_context->ctx=nvmpi_create_decoder(codectype,NV_PIX_YUV420);
In the patch, it is hardcoded to use YUV420. Is there a reason why this is preferred over NV12?

jocover, 2019-11-08:
ffmpeg tools decode video using yuv420,so I only did yuv420 support
if you want to using nv12 
please change https://github.com/jocover/jetson-ffmpeg/blob/master/ffmpeg_nvmpi.patch#L303
avctx->pix_fmt to AV_PIX_FMT_NV12
frame->format=AV_PIX_FMT_NV12;
(not tested)

xsacha, 2020-02-14:
This gives me:
"NvMMLiteOpen : Block : BlockType = 261 
NVMEDIA: Reading vendor.tegra.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 261 
Assertion ((src_linesize) >= 0 ? (src_linesize) : (-(src_linesize))) >= bytewidth failed at libavutil/imgutils.c:314"