jocover/jetson-ffmpeg #12: Static build

johnpdowling, 2019-11-05:
Is there any way to get this to build/install a static library version as well? Or any pointers for how-to? Trying to create a static build of ffmpeg and running into lots of pkg-config --static errors. Thanks!

st7109, 2019-11-07:
you should copy nvmpi.pc to /usr/local/lib/pkgconfig, then run below command:
export PKG_CONFIG_PATH="/usr/local/lib/pkgconfig"
I think you can get static lib.

martinhering, 2019-11-13:
In the CMakeLists.txt file, change the following line:

`add_library(nvmpi SHARED` to `add_library(nvmpi STATIC`

and comment out the install lines at the bottom. That did the trick for me.

johnpdowling, 2020-02-19:
Still nothing for me. I changed the lines in CMakeLists.txt, moved the resultant headers and libs to where they're referred to in the .pc file, and moved and put in the proper PATH flags. Still nothing, still "ERROR: nvmpi not found using pkg-config". Now there are errors in ffbuild/config.log whining about uclibc? I've also done a apt build-dep ffmpeg to see if there were other environmental factors, no change.

gjrtimmer, 2024-02-16:
> Still nothing for me. I changed the lines in CMakeLists.txt, moved the resultant headers and libs to where they're referred to in the .pc file, and moved and put in the proper PATH flags. Still nothing, still "ERROR: nvmpi not found using pkg-config" even though pkg-config --list-all and --list-all --static shows it. Now there are errors in ffbuild/config.log whining about uclibc? I've also done a apt build-dep ffmpeg to see if there were other environmental factors, no change.

Hi, I'm trying to build statically, could you give some more insight in what I need to change to create a static build.