jocover/jetson-ffmpeg #13: Crash when input is from rtsp

antinucleon, 2019-11-11:
Hi,

I am trying to use with https://github.com/KhaosT/homebridge-camera-ffmpeg. However seems patched ffmpeg is not able to recognize ioctl 

Log attached.
```
Input #0, rtsp, from 'rtsp://xxxxx':
  Metadata:
    title           : Media Presentation
  Duration: N/A, start: 0.299667, bitrate: N/A
    Stream #0:0: Video: h264 (Main), yuvj420p(pc, bt709, progressive), 1280x720 [SAR 1:1 DAR 16:9], 30 fps, 30 tbr, 90k tbn, 60 tbc
Failed to query video capabilities: Inappropriate ioctl for device
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 261 
NVMEDIA: Reading sys.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 261 
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (h264_nvmpi) -> wrapped_avframe (native))
Press [q] to stop, [?] for help
NVMAP_IOC_READ failed: Interrupted system call
NVMAP_IOC_READ: Offset 0 SrcStride 1280 pDst 0x7f84006330 DstStride 1280 Count 360
NVMAP_IOC_READ failed: Interrupted system call
NVMAP_IOC_READ: Offset 1048576 SrcStride 768 pDst 0x7f84006350 DstStride 640 Count 360
NVMAP_IOC_READ failed: Invalid argument
NVMAP_IOC_READ: Offset 1441792 SrcStride 768 pDst 0x7f84006370 DstStride 640 Count 1280
NVMAP_IOC_READ failed: Interrupted system call
NVMAP_IOC_READ: Offset 0 SrcStride 1280 pDst 0x7f84007e00 DstStride 1280 Count 360
NVMAP_IOC_READ failed: Interrupted system call
NVMAP_IOC_READ: Offset 1048576 SrcStride 768 pDst 0x7f84007e20 DstStride 640 Count 360
NVMAP_IOC_READ failed: Invalid argument
NVMAP_IOC_READ: Offset 1441792 SrcStride 768 pDst 0x7f84007e40 DstStride 640 Count 1280
Segmentation fault (core dumped)
```

jocover, 2019-11-12:
what's l4t version
please run
`dpkg-query --showformat='${Version}' --show nvidia-l4t-core`

antinucleon, 2019-11-14:
After install latest sdk it is solved and works well. Thanks!

closed 2019-11-14 by antinucleon