jocover/jetson-ffmpeg #14: How to verify if installed correctly

Tsopic, 2019-11-14:
I'm not sure if everything is correct. 

I'm not able to use the acceleration and want to know if there is a way to check if installion is done correctly.

my `ffmpeg -hwaccels`

```
ffmpeg -hwaccels
ffmpeg version 3.4.6-0ubuntu0.18.04.1 Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 7 (Ubuntu/Linaro 7.3.0-16ubuntu3)
  configuration: --prefix=/usr --extra-version=0ubuntu0.18.04.1 --toolchain=hardened --libdir=/usr/lib/aarch64-linux-gnu --incdir=/usr/include/aarch64-linux-gnu --enable-gpl --disable-stripping --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librubberband --enable-librsvg --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libopencv --enable-libx264 --enable-shared
  libavutil      55. 78.100 / 55. 78.100
  libavcodec     57.107.100 / 57.107.100
  libavformat    57. 83.100 / 57. 83.100
  libavdevice    57. 10.100 / 57. 10.100
  libavfilter     6.107.100 /  6.107.100
  libavresample   3.  7.  0 /  3.  7.  0
  libswscale      4.  8.100 /  4.  8.100
  libswresample   2.  9.100 /  2.  9.100
  libpostproc    54.  7.100 / 54.  7.100
Hardware acceleration methods:
vdpau
vaapi
```

jocover, 2019-11-15:
not support ffmpeg hardware acceleration now.
because ffmpeg v4l2 request api not working now, need more test

Tsopic, 2019-11-15:
Would it work with older ffmpeg version?

If that's the case, which version?

closed 2021-08-13 by Tsopic

cizekmilan, 2021-08-18:
hello can i join here?
Can anyone provide what looks like the load of his GPU utilization in jtop?

I have two test of transcoding, both examples GPUs use little. (only encoding give me the same poor results)
`ffmpeg -y -benchmark -c:v h264_nvmpi -i "./samples/CT-Sport-AVC-1080p50-v.ts" -c:v hevc_nvmpi test_hevc.ts`
`ffmpeg -y -benchmark -c:v hevc_nvmpi -i "./samples/CT24-HEVC-1080p50-v.ts" -c:v h264_nvmpi test_h264.ts`

https://snipboard.io/m5B3xv.jpg
https://snipboard.io/m6LOzP.jpg
https://snipboard.io/1xlSK8.jpg

Is it normal? I'm disappointed with too low performance on my Jetson Nano.

`ffmpeg version c9f3835 Copyright (c) 2000-2020 the FFmpeg developers
  built with gcc 7 (Ubuntu/Linaro 7.5.0-3ubuntu1~18.04)
  configuration: --enable-static --pkg-config-flags=--static --disable-shared --prefix=/tmp/stream_install/ffmpeg-build-static --extra-cflags='-I /tmp/stream_install/ffmpeg-build-static/include -I /usr/local/cuda/include/' --extra-ldflags='-L /tmp/stream_install/ffmpeg-build-static/lib -L /usr/local/cuda/lib64/' --enable-gpl --enable-nonfree --enable-libfdk-aac --enable-nvmpi --enable-cuda-nvcc --enable-libnpp --enable-pthreads --enable-runtime-cpudetect --enable-filter=drawtext --enable-libfreetype`

Inputs:
`# mediainfo CT-Sport-AVC-1080p50-v.ts
General
ID                                       : 1 (0x1)
Complete name                            : CT-Sport-AVC-1080p50-v.ts
Format                                   : MPEG-TS
File size                                : 258 MiB
Duration                                 : 4 min 58 s
Overall bit rate mode                    : Variable
Overall bit rate                         : 7 244 kb/s

Video
ID                                       : 256 (0x100)
Menu ID                                  : 1 (0x1)
Format                                   : AVC
Format/Info                              : Advanced Video Codec
Format profile                           : High@L4.2
Format settings                          : CABAC / 4 Ref Frames
Format settings, CABAC                   : Yes
Format settings, Reference frames        : 4 frames
Codec ID                                 : 27
Duration                                 : 4 min 58 s
Bit rate                                 : 6 886 kb/s
Width                                    : 1 920 pixels
Height                                   : 1 080 pixels
Display aspect ratio                     : 16:9
Frame rate mode                          : Variable
Color space                              : YUV
Chroma subsampling                       : 4:2:0
Bit depth                                : 8 bits
Scan type                                : Progressive
Stream size                              : 245 MiB (95%)
Color range                              : Limited
Color primaries                          : BT.709
Transfer characteristics                 : BT.709
Matrix coefficients                      : BT.709

Menu
ID                                       : 4096 (0x1000)
Menu ID                                  : 1 (0x1)
Duration                                 : 4 min 58 s
List                                     : 256 (0x100) (AVC)
Service name                             : Service01
Service provider                         : FFmpeg
Service type                             : digital television`

`# mediainfo CT24-HEVC-1080p50-v.ts
General
ID                                       : 1 (0x1)
Complete name                            : CT24-HEVC-1080p50-v.ts
Format                                   : MPEG-TS
File size                                : 141 MiB
Duration                                 : 4 min 58 s
Overall bit rate mode                    : Variable
Overall bit rate                         : 3 962 kb/s

Video
ID                                       : 256 (0x100)
Menu ID                                  : 1 (0x1)
Format                                   : HEVC
Format/Info                              : High Efficiency Video Coding
Format profile                           : Main@L4.1@Main
Codec ID                                 : 36
Duration                                 : 4 min 58 s
Bit rate                                 : 3 767 kb/s
Width                                    : 1 920 pixels
Height                                   : 1 080 pixels
Display aspect ratio                     : 16:9
Frame rate                               : 50.000 FPS
Color space                              : YUV
Chroma subsampling                       : 4:2:0
Bit depth                                : 8 bits
Bits/(Pixel*Frame)                       : 0.036
Stream size                              : 134 MiB (95%)
Color range                              : Limited
Color primaries                          : BT.709
Transfer characteristics                 : BT.709
Matrix coefficients                      : BT.709

Menu
ID                                       : 4096 (0x1000)
Menu ID                                  : 1 (0x1)
Duration                                 : 4 min 58 s
List                                     : 256 (0x100) (HEVC)
Service name                             : Service01
Service provider                         : FFmpeg
Service type                             : digital television
`

grantthomas, 2021-08-18:
> hello can i join here?
It'd probably be recommended to start a new issue / question instead of adding to this one that is closed.

> Can anyone provide what looks like the load of his GPU utilization in jtop?
In your second picture, the NVENC and NVDEC are showing as running.
https://snipboard.io/m6LOzP.jpg

The NVENC and NVDEC usage does not show up on the GPU view of jtop.
Your ffmpeg is working, and you're using hardware acceleration for both decode and encode.

> Is it normal? I'm disappointed with too low performance on my Jetson Nano.
What specifically are you disappointed with, and what were you expecting?
framerate? encode quality? something else?

cizekmilan, 2021-08-19:
I have one more question. It seems to me that the bitrate setting is ignored, compared to other codecs. I define e.g.:
to HEVC: `-profile:v main -level:v 4.1 -minrate 4000k -b:v 4500k -maxrate 6000k`
to H264: `-profile: v high -level: v 4 -minrate 4000k -b: v 4500k -maxrate 6000k`

But the resulting bitrate averages around 8Mbit. The resulting file is, for example, 2x larger (both H.264 and HEVC). This is how the file size comparison based on codecs with the above defined parameters looks like: https://snipboard.io/h8y5De.jpg (you can see the size difference). Thank you.