jocover/jetson-ffmpeg #17: Segmentation fault (core dumped)

heshunchao, 2019-11-28:
root@heshunchao-desktop:/home/ffmpeg# ./ffmpeg -c:v h264_nvmpi -i 1.264 -f null -
ffmpeg version 82a3a62 Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 7 (Ubuntu/Linaro 7.4.0-1ubuntu1~18.04.1)
  configuration: --enable-nvmpi
  libavutil      56. 31.100 / 56. 31.100
  libavcodec     58. 54.100 / 58. 54.100
  libavformat    58. 29.100 / 58. 29.100
  libavdevice    58.  8.100 / 58.  8.100
  libavfilter     7. 57.100 /  7. 57.100
  libswscale      5.  5.100 /  5.  5.100
  libswresample   3.  5.100 /  3.  5.100
[h264 @ 0x5581d15570] SEI type 128 size 576 truncated at 128
[h264 @ 0x5581d15570] SEI type 128 size 576 truncated at 95
[h264 @ 0x5581d15570] SEI type 128 size 576 truncated at 128
    Last message repeated 2 times
Input #0, h264, from '1.264':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: h264 (High), yuvj420p(pc, bt709, progressive), 1920x1080, 25 fps, 25 tbr, 1200k tbn, 50 tbc
Failed to query video capabilities: Inappropriate ioctl for device
Opening in BLOCKING MODE
NvMMLiteOpen : Block : BlockType = 261
NVMEDIA: Reading sys.display-size : status: 6
NvMMLiteBlockCreate : Block : BlockType = 261
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (h264_nvmpi) -> wrapped_avframe (native))
Press [q] to stop, [?] for help
NVMAP_IOC_READ failed: Interrupted system call
NVMAP_IOC_READ: Offset 0 SrcStride 2048 pDst 0x7f84005dc0 DstStride 1920 Count 540
NVMAP_IOC_READ failed: Interrupted system call
NVMAP_IOC_READ: Offset 2228224 SrcStride 1024 pDst 0x7f84005de0 DstStride 960 Count 540
NVMAP_IOC_READ failed: Invalid argument
NVMAP_IOC_READ: Offset 2883584 SrcStride 1024 pDst 0x7f84005e00 DstStride 960 Count 2048
NVMAP_IOC_READ failed: Interrupted system call
NVMAP_IOC_READ: Offset 0 SrcStride 2048 pDst 0x7f84007890 DstStride 1920 Count 540
NVMAP_IOC_READ failed: Interrupted system call
NVMAP_IOC_READ: Offset 2228224 SrcStride 1024 pDst 0x7f840078b0 DstStride 960 Count 540
NVMAP_IOC_READ failed: Invalid argument
NVMAP_IOC_READ: Offset 2883584 SrcStride 1024 pDst 0x7f840078d0 DstStride 960 Count 2048
Segmentation fault (core dumped)

How to deal with this bug?

jocover, 2019-11-28:
what's you l4t version

heshunchao, 2019-11-29:
# R32 (release), REVISION: 1.0, GCID: 14531094, BOARD: t210ref, EABI: aarch64, DATE: Wed Mar 13 07:46:13 UTC 2019

jocover, 2019-11-29:
please update to the latest version,this work on 32.2.1 or latest

heshunchao, 2019-11-30:
Ok， thank you very much

heshunchao, 2019-12-01:
We have tested and succeeded, thanks again