jocover/jetson-ffmpeg #21: ffplay with rtsp stream

uutzinger, 2019-12-18:
I compile with 
--enable-nvmpi --enable-nonfree --enable-shared --enable-ffplay 
and I installed libsdl2 with "apt install libsdl2-dev " to get ffplay built.

When running "ffplay -rtsp_transport udp rtsp://192.168.11.202:8554/unicast"

I get 
dbus[29617]: arguments to dbus_message_new_method_call() were incorrect, assertion "path != NULL" failed in file ../../../dbus/dbus-message.c line 1362.
This is normally a bug in some application using the D-Bus library.

I make sure I can play the stream with gstreamer:
gst-launch-1.0 rtspsrc location="rtsp://192.168.11.202:8554/unicast" latency=100 ! rtph264depay ! h264parse ! omxh264dec ! nvoverlaysink overlay-x=0 overlay-y=20 overlay-w=960 overlay-h=540 overlay=2

What am I missing?

RicoKirstein, 2020-01-06:
try
DBUS_FATAL_WARNINGS=0 ffplay -rtsp_transport udp rtsp://192.168.11.202:8554/unicast
or
export DBUS_FATAL_WARNINGS=0 
and then run your program
i had a similar problem with it when trying to use in moonlight

PTS93, 2020-03-10:
this is a bug with sdl2 and dbug in ubuntu 18.04
you can temporarily also kill ibus by running sudo killall ibus-daemon