jocover/jetson-ffmpeg #30: Missing license

cgutman, 2020-02-26:
I noticed this code is missing a license. Can you please add a license file to the repository so the code can be legally redistributed?

As stated on https://choosealicense.com/no-permission/:
> If you find software that doesn’t have a license, that generally means you have no permission from the creators of the software to use, modify, or share the software. Although a code host such as GitHub may allow you to view and fork the code, this does not imply that you are permitted to use, modify, or share the software for any purpose.

jocover, 2020-02-26:
Thank you for your advice,I added MIT License

cgutman, 2020-02-28:
Thanks, looks good.

closed 2020-02-28 by cgutman