jocover/jetson-ffmpeg #31: could not play main profile stream from Hikvision

bellkin, 2020-02-28:
Both stream are recorded from Hikvision DS-2CD3955FWD-IWS(not start with iframe)
1. for H264, gstreamer play rtsp with heavy frame drop, probably timing bug, but play file is fine.
2. for HEVC stream, gstreamer play both rtsp/file fine.
https://www.mediafire.com/folder/8lq8sdsixr1yd/

bellkin, 2020-02-29:
Conditions tested:
1. Not work with any resolution: 1280x960,2048x1536,2560x1920.
2. Not work with color space change to YUV420:
perl -pe 's/\x{CF}\x{F8}\x{0B}\x{70}/\x{CF}\x{F8}\x{0B}\x{50}/g' < h264er.mp4 > h264nr.mp4
3. Not work with NV12 output.
error message:
NvDdkVicConfigure Failed
nvbuffer_transform Failed

st7109, 2020-04-27:
should add support about AV_PIX_FMT_YUVJ420P...just same as AV_PIX_FMT_YUV420P

bellkin, 2020-05-13:
fixed by simply update header files as:
https://github.com/jocover/jetson-ffmpeg/issues/32#issuecomment-625543955

closed 2020-05-13 by bellkin