jocover/jetson-ffmpeg #32: "nvbuffer_transform Failed" when decoding with h264_nvmpi

PTS93, 2020-03-10:
When trying to decode a video with ffplay I get the following error on a 4k h264 video.
I also attached a trace log if that helps: [ffplay.log](https://github.com/jocover/jetson-ffmpeg/files/4312922/ffplay.log)
I used the following command: `./ffplay feather_reflow-4K.mp4 -codec:v h264_nvmpi`

Running L4T 32.3.1 on Jetson Nano.

```
Transform failed
NvDdkVicConfigure Failed
nvbuffer_transform Failed
```

PTS93, 2020-03-10:
Same issue when running the decoding `ffmpeg -c:v h264_nvmpi -i input_file -f null -` with the same file. Same with a 720p h264 file (main profile)

8SuperMaks8, 2020-04-15:
I have the same issue, but in my case, Im trying to decode 4k hevc_nvmpi.
The output for the short video:
`...
[AVBSFContext @ 0x557f3ac8d0] No parameter sets in the extradata
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 279 
NVMEDIA: Reading vendor.tegra.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 279 
Stream mapping:
  Stream #0:0 -> #0:0 (hevc (hevc_nvmpi) -> wrapped_avframe (native))
  Stream #0:1 -> #0:1 (eac3 (native) -> pcm_s16le (native))
Press [q] to stop, [?] for help
NvDdkVicConfigure Failed
nvbuffer_transform Failed
Transform failedNvDdkVicConfigure Failed
nvbuffer_transform Failed
Transform failedNvDdkVicConfigure Failed
nvbuffer_transform Failed
Transform failedNvDdkVicConfigure Failed
nvbuffer_transform Failed
Assertion ((src_linesize) >= 0 ? (src_linesize) : (-(src_linesize))) >= bytewidth failed at src/libavutil/imgutils.c:314
Aborted (core dumped)`

And for the long 4k hevec movie:
`....
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 279 
NVMEDIA: Reading vendor.tegra.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 279 
Stream mapping:
  Stream #0:0 -> #0:0 (hevc (hevc_nvmpi) -> wrapped_avframe (native))
  Stream #0:1 -> #0:1 (ac3 (native) -> pcm_s16le (native))
Press [q] to stop, [?] for help
Segmentation fault (core dumped)`

dl-nobrolla, 2020-05-07:
I had this same probe, the fix is to use nvbuf_utils.h from 'Jetson Linux Multimedia API' and not the one in the include directory in this repo.  The NvBufferColorFormat enum is different, nvidia must have updated this in a way that's not backwards compatable.