jocover/jetson-ffmpeg #34: cmake .. fails

galasso9423, 2020-04-07:
galasso@jet2:~/jetson-ffmpeg/build$ cmake ..
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LIB_V4L2
    linked by target "nvmpi" in directory /home/galasso/jetson-ffmpeg

jocover, 2020-04-07:
try to using this cmake file on jetpack 4.3 
https://gist.github.com/jocover/7a60c1006e546ed821d4e609d80df765

galasso9423, 2020-04-07:
Thank you. It worked

closed 2020-04-07 by galasso9423