jocover/jetson-ffmpeg #37: Can't build due to error with LIB_V4L2

Tombert, 2020-04-26:
Hello, 
When I try and go through the build steps with cmake, I get this error: 

```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LIB_V4L2
    linked by target "nvmpi" in directory /home/tombert/jetson-ffmpeg

-- Configuring incomplete, errors occurred!
See also "/home/tombert/jetson-ffmpeg/build/CMakeFiles/CMakeOutput.log".
See also "/home/tombert/jetson-ffmpeg/build/CMakeFiles/CMakeError.log".

```

Is there something that I'm obviously missing?

jocover, 2020-04-27:
try to using this cmake file on jetpack 4.3
https://gist.github.com/jocover/7a60c1006e546ed821d4e609d80df765

Tombert, 2020-04-27:
That worked..thanks!

closed 2020-04-27 by Tombert

7oud, 2020-04-28:
@jocover  
could you paste the content of cmake file here, I can not open the link for some resason

jocover, 2020-04-28:
cmake_minimum_required(VERSION 3.9)
project(nvmpi VERSION 1.0.0 DESCRIPTION "nvidia multimedia api")

set(CMAKE_C_FLAGS“$ {CMAKE_C_FLAGS} -fPIC”)
set(CMAKE_CXX_FLAGS“$ {CMAKE_CXX_FLAGS} -fPIC”)
#set(CMAKE_SHARED_LINKER_FLAGS "${CMAKE_SHARED_LINKER_FLAGS}
-Wl,-rpath-link=/lib")
#set(CMAKE_SHARED_LINKER_FLAGS "${CMAKE_SHARED_LINKER_FLAGS}
-Wl,-rpath-link=/usr/lib/aarch64-linux-gnu")
set(CMAKE_SHARED_LINKER_FLAGS "${CMAKE_SHARED_LINKER_FLAGS}
-Wl,-rpath-link=/usr/lib/aarch64-linux-gnu/tegra")
set(CMAKE_SHARED_LINKER_FLAGS "${CMAKE_SHARED_LINKER_FLAGS}
-Wl,-rpath-link=/usr/local/cuda/lib64")

find_library(LIB_NVBUF nvbuf_utils PATHS /usr/lib/aarch64-linux-gnu/tegra)
find_library(LIB_V4L2 nvv4l2 PATHS /usr/lib/aarch64-linux-gnu/tegra)
find_package (Threads)
#find_library(LIB_DRM drm PATHS /usr/lib/aarch64-linux-gnu/tegra)
#find_library(LIB_EGL EGL PATHS /usr/lib/aarch64-linux-gnu/tegra)

add_library(nvmpi SHARED
    nvmpi_dec.cpp
    nvmpi_enc.cpp
    /usr/src/jetson_multimedia_api/samples/common/classes/NvBuffer.cpp
    /usr/src/jetson_multimedia_api/samples/common/classes/NvElement.cpp

/usr/src/jetson_multimedia_api/samples/common/classes/NvElementProfiler.cpp
    /usr/src/jetson_multimedia_api/samples/common/classes/NvLogging.cpp
    /usr/src/jetson_multimedia_api/samples/common/classes/NvV4l2Element.cpp

/usr/src/jetson_multimedia_api/samples/common/classes/NvV4l2ElementPlane.cpp
    /usr/src/jetson_multimedia_api/samples/common/classes/NvVideoDecoder.cpp
    /usr/src/jetson_multimedia_api/samples/common/classes/NvVideoEncoder.cpp
    #common/NvVideoConverter.cpp
    #common/NvApplicationProfiler.cpp
    #common/NvEglRenderer.cpp
    #common/NvUtils.cpp
)

set_target_properties(nvmpi PROPERTIES VERSION ${PROJECT_VERSION})
set_target_properties(nvmpi PROPERTIES PUBLIC_HEADER nvmpi.h)
target_link_libraries(nvmpi PRIVATE ${CMAKE_THREAD_LIBS_INIT} ${LIB_NVBUF}
 ${LIB_V4L2})
target_include_directories(nvmpi PRIVATE
/usr/src/jetson_multimedia_api/include)
target_include_directories(nvmpi PRIVATE /usr/local/cuda/include)
configure_file(nvmpi.pc.in nvmpi.pc @ONLY)
include(GNUInstallDirs)
install(TARGETS ${PROJECT_NAME}
   LIBRARY DESTINATION ${CMAKE_INSTALL_LIBDIR}
       PUBLIC_HEADER DESTINATION ${CMAKE_INSTALL_INCLUDEDIR})
install(FILES ${CMAKE_BINARY_DIR}/nvmpi.pc
   DESTINATION ${CMAKE_INSTALL_DATAROOTDIR}/pkgconfig)

>

razvanphp, 2020-05-30:
this still triggers this warning:
```
jetson@jetson:~/jetson-ffmpeg/build$ cmake ..
CMake Warning (dev) at CMakeLists.txt:39 (configure_file):
  configure_file called with unknown argument(s):

   @only

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done
CMake Warning at CMakeLists.txt:17 (add_library):
  Cannot generate a safe runtime search path for target nvmpi because files
  in some directories may conflict with libraries in implicit directories:

    runtime library [libv4l2.so.0] in /usr/lib/aarch64-linux-gnu may be hidden by files in:
      /usr/lib/aarch64-linux-gnu/tegra

  Some of these libraries may not be found correctly.

-- Generating done
-- Build files have been written to: /home/jetson/jetson-ffmpeg/build
jetson@jetson:~/jetson-ffmpeg/build$
```
also, when compiling ffmpeg 4.2 with `--enable-shared` then the `nvmpi` decoder does not show up in the `-encoders` list. Any hints?

razvanphp, 2020-05-30:
sudo mv /usr/local/lib/libnvmpi.so* /usr/lib/aarch64-linux-gnu/
sudo mv /usr/local/include/nvmpi.h /usr/include/aarch64-linux-gnu/