jocover/jetson-ffmpeg #38: Decoding MPEG2 results in a segmentation fault

Tombert, 2020-04-27:
Hello!

I was trying to decode a DVD rip I made using this command: `ffmpeg -c:v mpeg2_nvmpi -i title_t00.mkv -c:v hevc_nvmpi -c:a flac boondocks_01.mkv`, but this is the result I got: 

```
tombert@jetson1:/tank/boondocks_disc_1$ ffmpeg -c:v mpeg2_nvmpi -i title_t00.mkv -c:v hevc_nvmpi -c:a flac boondocks_01.mkv
ffmpeg version d1e8be3 Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 7 (Ubuntu/Linaro 7.5.0-3ubuntu1~18.04)
  configuration: --enable-nvmpi
  libavutil      56. 31.100 / 56. 31.100
  libavcodec     58. 54.100 / 58. 54.100
  libavformat    58. 29.100 / 58. 29.100
  libavdevice    58.  8.100 / 58.  8.100
  libavfilter     7. 57.100 /  7. 57.100
  libswscale      5.  5.100 /  5.  5.100
  libswresample   3.  5.100 /  3.  5.100
Input #0, matroska,webm, from 'title_t00.mkv':
  Metadata:
    encoder         : libmakemkv v1.14.5 (1.3.5/1.4.7) x86_64-linux-gnu
    creation_time   : 2020-04-21T00:08:33.000000Z
  Duration: 00:19:58.16, start: 0.000000, bitrate: 5303 kb/s
    Chapter #0:0: start 0.000000, end 105.972533
    Metadata:
      title           : Chapter 01
    Chapter #0:1: start 105.972533, end 559.458900
    Metadata:
      title           : Chapter 02
    Chapter #0:2: start 559.458900, end 1165.631133
    Metadata:
      title           : Chapter 03
    Chapter #0:3: start 1165.631133, end 1196.662133
    Metadata:
      title           : Chapter 04
    Chapter #0:4: start 1196.662133, end 1198.163633
    Metadata:
      title           : Chapter 05
    Stream #0:0(eng): Video: mpeg2video (Main), yuv420p(tv, top first), 720x480 [SAR 32:27 DAR 16:9], SAR 186:157 DAR 279:157, 29.97 fps, 29.97 tbr, 1k tbn, 59.94 tbc
    Metadata:
      BPS-eng         : 4904729
      DURATION-eng    : 00:19:58.163633333
      NUMBER_OF_FRAMES-eng: 35909
      NUMBER_OF_BYTES-eng: 734583144
      SOURCE_ID-eng   : 0100E0
      _STATISTICS_WRITING_APP-eng: MakeMKV v1.14.5 linux(x64-release)
      _STATISTICS_WRITING_DATE_UTC-eng: 2020-04-21 00:08:33
      _STATISTICS_TAGS-eng: BPS DURATION NUMBER_OF_FRAMES NUMBER_OF_BYTES SOURCE_ID
    Stream #0:1(eng): Audio: ac3, 48000 Hz, stereo, fltp, 192 kb/s (default)
    Metadata:
      title           : Stereo
      BPS-eng         : 192000
      DURATION-eng    : 00:19:58.144000000
      NUMBER_OF_FRAMES-eng: 37442
      NUMBER_OF_BYTES-eng: 28755456
      SOURCE_ID-eng   : 0180BD
      _STATISTICS_WRITING_APP-eng: MakeMKV v1.14.5 linux(x64-release)
      _STATISTICS_WRITING_DATE_UTC-eng: 2020-04-21 00:08:33
      _STATISTICS_TAGS-eng: BPS DURATION NUMBER_OF_FRAMES NUMBER_OF_BYTES SOURCE_ID
    Stream #0:2(eng): Audio: ac3, 48000 Hz, stereo, fltp, 192 kb/s
    Metadata:
      title           : Stereo
      BPS-eng         : 192000
      DURATION-eng    : 00:19:58.144000000
      NUMBER_OF_FRAMES-eng: 37442
      NUMBER_OF_BYTES-eng: 28755456
      SOURCE_ID-eng   : 0181BD
      _STATISTICS_WRITING_APP-eng: MakeMKV v1.14.5 linux(x64-release)
      _STATISTICS_WRITING_DATE_UTC-eng: 2020-04-21 00:08:33
      _STATISTICS_TAGS-eng: BPS DURATION NUMBER_OF_FRAMES NUMBER_OF_BYTES SOURCE_ID
    Stream #0:3(fre): Subtitle: dvd_subtitle, 720x480 (default)
    Metadata:
      BPS-eng         : 7700
      DURATION-eng    : 00:19:53.058255555
      NUMBER_OF_FRAMES-eng: 318
      NUMBER_OF_BYTES-eng: 1148322
      SOURCE_ID-eng   : 0120BD
      _STATISTICS_WRITING_APP-eng: MakeMKV v1.14.5 linux(x64-release)
      _STATISTICS_WRITING_DATE_UTC-eng: 2020-04-21 00:08:33
      _STATISTICS_TAGS-eng: BPS DURATION NUMBER_OF_FRAMES NUMBER_OF_BYTES SOURCE_ID
    Stream #0:4(eng): Subtitle: subrip
    Metadata:
      BPS-eng         : 166
      DURATION-eng    : 00:19:18.688000000
      NUMBER_OF_FRAMES-eng: 477
      NUMBER_OF_BYTES-eng: 24137
      SOURCE_ID-eng   : 0100E0
      _STATISTICS_WRITING_APP-eng: MakeMKV v1.14.5 linux(x64-release)
      _STATISTICS_WRITING_DATE_UTC-eng: 2020-04-21 00:08:33
      _STATISTICS_TAGS-eng: BPS DURATION NUMBER_OF_FRAMES NUMBER_OF_BYTES SOURCE_ID
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 267 
NVMEDIA: Reading vendor.tegra.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 267 
Stream mapping:
  Stream #0:0 -> #0:0 (mpeg2video (mpeg2_nvmpi) -> hevc (hevc_nvmpi))
  Stream #0:1 -> #0:1 (ac3 (native) -> flac (native))
  Stream #0:4 -> #0:2 (subrip (srt) -> ass (ssa))
Press [q] to stop, [?] for help
[flac @ 0x558404dac0] encoding as 24 bits-per-sample
NVMEDIA: NvMediaMixerInit: 119: frameWidth = 720, frameHeight = 480 
NVMEDIA: DeinterlaceThread: 782: DeinterlaceThread is created 
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 8 
===== NVMEDIA: NVENC =====
NvMMLiteBlockCreate : Block : BlockType = 8 
892744264
842091865
Output #0, matroska, to 'boondocks_01.mkv':
  Metadata:
    encoder         : Lavf58.29.100
    Chapter #0:0: start 0.000000, end 105.972533
    Metadata:
      title           : Chapter 01
    Chapter #0:1: start 105.972533, end 559.458900
    Metadata:
      title           : Chapter 02
    Chapter #0:2: start 559.458900, end 1165.631133
    Metadata:
      title           : Chapter 03
    Chapter #0:3: start 1165.631133, end 1196.662133
    Metadata:
      title           : Chapter 04
    Chapter #0:4: start 1196.662133, end 1198.163633
    Metadata:
      title           : Chapter 05
    Stream #0:0(eng): Video: hevc (hevc_nvmpi), yuv420p, 720x480 [SAR 186:157 DAR 279:157], q=-1--1, 2000 kb/s, 29.97 fps, 1k tbn, 29.97 tbc
    Metadata:
      BPS-eng         : 4904729
      DURATION-eng    : 00:19:58.163633333
      NUMBER_OF_FRAMES-eng: 35909
      NUMBER_OF_BYTES-eng: 734583144
      SOURCE_ID-eng   : 0100E0
      _STATISTICS_WRITING_APP-eng: MakeMKV v1.14.5 linux(x64-release)
      _STATISTICS_WRITING_DATE_UTC-eng: 2020-04-21 00:08:33
      _STATISTICS_TAGS-eng: BPS DURATION NUMBER_OF_FRAMES NUMBER_OF_BYTES SOURCE_ID
      encoder         : Lavc58.54.100 hevc_nvmpi
    Stream #0:1(eng): Audio: flac ([172][241][0][0] / 0xF1AC), 48000 Hz, stereo, s32 (24 bit), 128 kb/s (default)
    Metadata:
      title           : Stereo
      BPS-eng         : 192000
      DURATION-eng    : 00:19:58.144000000
      NUMBER_OF_FRAMES-eng: 37442
      NUMBER_OF_BYTES-eng: 28755456
      SOURCE_ID-eng   : 0180BD
      _STATISTICS_WRITING_APP-eng: MakeMKV v1.14.5 linux(x64-release)
      _STATISTICS_WRITING_DATE_UTC-eng: 2020-04-21 00:08:33
      _STATISTICS_TAGS-eng: BPS DURATION NUMBER_OF_FRAMES NUMBER_OF_BYTES SOURCE_ID
      encoder         : Lavc58.54.100 flac
    Stream #0:2(eng): Subtitle: ass (ssa)
    Metadata:
      BPS-eng         : 166
      DURATION-eng    : 00:19:18.688000000
      NUMBER_OF_FRAMES-eng: 477
      NUMBER_OF_BYTES-eng: 24137
      SOURCE_ID-eng   : 0100E0
      _STATISTICS_WRITING_APP-eng: MakeMKV v1.14.5 linux(x64-release)
      _STATISTICS_WRITING_DATE_UTC-eng: 2020-04-21 00:08:33
      _STATISTICS_TAGS-eng: BPS DURATION NUMBER_OF_FRAMES NUMBER_OF_BYTES SOURCE_ID
      encoder         : Lavc58.54.100 ssa
NVMEDIA: H265 : Profile : 3 
Segmentation fault (core dumped)

```

This also happens if I omit the hardware decoder (`-c:v mpeg2_nvmpi`) and just go straight to the `-i`

Is there something I'm missing or doing wrong?

Tombert, 2020-04-27:
Just to confirm that I'm not too off base with the codec, I ran this command: 

```
$ ffprobe -v error -select_streams v:0 -show_entries stream=codec_name -of default=noprint_wrappers=1:nokey=1 title_t00.mkv 
mpeg2video
```

aspecialkey, 2020-08-07:
I ran into the same problem. Is there a soltion? (I`m new to jetso nano - use the default sd-card-image and follow the described install instruction for ffmpeg)

icebear74, 2020-08-19:
Hi,
Same here .. Is there a solution ? or can i force ffmpeg to NOT user jetson to decode the mpeg2video ?

syadnom, 2021-01-05:
If you omit anything to the left of the -i then it should be software decoded...  Try and it see if your CPU gets slammed

GavinDarkglider, 2021-11-02:
This needs renamed, as the mpeg2 decoder is actually starting and working. You need to look closer at that log, or I guess know what you are looking at.
Decoder opening....
[code]
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 267 
NVMEDIA: Reading vendor.tegra.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 267 
[/code]
Encoder opening....
[code]
NVMEDIA: NvMediaMixerInit: 119: frameWidth = 720, frameHeight = 480 
NVMEDIA: DeinterlaceThread: 782: DeinterlaceThread is created 
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 8 
===== NVMEDIA: NVENC =====
NvMMLiteBlockCreate : Block : BlockType = 8 
[/code]
As for the segfault, that could be due to a number of things, and without further investigation, or the core dump, or gdb info on where the bug is.....