jocover/jetson-ffmpeg #41: Encoder can return correct packets with pts/dts < 1000000

teplofizik, 2020-05-26:
PTS/DTS return as PTS % 1000000

avcodec_send_frame(); <= pts: 1000123
avcodec_receive_packet() => pts: 123

Error:
`[flv @ 0x55ceac0a60] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 999999 >= 32`

teplofizik, 2020-05-27:
nvmpi_enc.cpp:
`415	v4l2_buf.timestamp.tv_usec = frame->timestamp;`
to
`415	v4l2_buf.timestamp.tv_usec = frame->timestamp % 1000000;
416	v4l2_buf.timestamp.tv_sec = frame->timestamp / 1000000;`

`88 ctx->timestamp[ctx->buf_index]=v4l2_buf->timestamp.tv_usec;`
to
`88 ctx->timestamp[ctx->buf_index] = (v4l2_buf->timestamp.tv_usec % 1000000) + (v4l2_buf->timestamp.tv_sec * 1000000UL);`

teplofizik, 2020-05-27:
[fix_nvmpi.txt](https://github.com/jocover/jetson-ffmpeg/files/4687409/fix_nvmpi.txt)

jocover, 2020-05-27:
I remember jetson nano nvenc does not support hardware timestamp, so I added V4L2_BUF_FLAG_TIMESTAMP_COPY flags

teplofizik, 2020-05-27:
I don't know about hardware timestamp. It is all about copying timestamp from input to output packet. tv_usec field cropped by system to fit in diapasone 0-999999... And tv_sec field at now not used at all.