jocover/jetson-ffmpeg #42: Not Utilizing NVENC

alexkOtto, 2020-05-31:
Hello,

As far as I understand, using the h264_nvmpi/hevc_nvmpi after patching and installing correctly is supposed to use nvenc for video encoding. I tried encoding a file using it and monitoring using jtop. I was surprised to see only cpu usage and no nvenc usage at all.

Is this OK or is there a bug? Maybe I did something wrong? I followed the instructions exactly.

cizekmilan, 2020-06-02:
I have the same problem. The application of the patch shows some errors.

jesusluque, 2020-06-05:
me too, only CPU with GPU 20% aprox 
sudo  jtop.. ...
no NVDEC (OFF) 
jetson nano.

jocover, 2020-06-05:
what is your JetPack version

cizekmilan, 2020-06-05:
I used nv-jetson-nano-sd-card-image-r32.4.2.zip image, then I did apt upgrade/update and then compile nvmpi and ffmpeg.

jesusluque, 2020-06-05:
Jeson 4.4 L4T 32.4.2

jocover, 2020-06-05:
added cmake patch
https://github.com/jocover/jetson-ffmpeg/commit/bda55bf84a7d8129975b0db23c9ac1db96485379

cizekmilan, 2020-06-05:
I'm going to try your update. I dont know what means git apply, but when I try "patch < ffmpeg_nvmpi.patch" (updated version), I see some fails, is it ok?

patching file configure
patching file Makefile
Hunk #1 FAILED at 354.
Hunk #2 FAILED at 379.
Hunk #3 FAILED at 464.
Hunk #4 FAILED at 669.
Hunk #5 FAILED at 679.
5 out of 5 hunks FAILED -- saving rejects to file Makefile.rej
can't find file to patch at input line 146
Perhaps you should have used the -p or --strip option?
The text leading up to this was:
--------------------------
|diff --git a/libavcodec/allcodecs.c b/libavcodec/allcodecs.c
|index d2f9a39..04dc62b 100644
|--- a/libavcodec/allcodecs.c
|+++ b/libavcodec/allcodecs.c
--------------------------
File to patch:

jocover, 2020-06-06:
> git apply seems OK
> 
> ```
> # git apply -v ffmpeg_nvmpi.patch
> Checking patch configure...
> Checking patch libavcodec/Makefile...
> Checking patch libavcodec/allcodecs.c...
> Checking patch libavcodec/nvmpi_dec.c...
> Checking patch libavcodec/nvmpi_enc.c...
> Applied patch configure cleanly.
> Applied patch libavcodec/Makefile cleanly.
> Applied patch libavcodec/allcodecs.c cleanly.
> Applied patch libavcodec/nvmpi_dec.c cleanly.
> Applied patch libavcodec/nvmpi_enc.c cleanly.
> ```
> 
> But problem stills. When I check encoding to h264 or hevc, only CPU is loaded (without GPU, checked with jtop) with low fps. My testing commands are:
> 
> ```
> ffmpeg -y -loglevel debug -f lavfi -i testsrc="size=1920x1080:rate=50" -t 10 -c:v h264_nvmpi /hls/test_h264.mp4
> ffmpeg -y -loglevel debug -f lavfi -i testsrc="size=1920x1080:rate=50" -t 10 -c:v hevc_nvmpi /hls/test_hevc.mp4
> ```

maybe RGB->YUV is too slow on jetson nano,this uses CPU conversion
`./ffmpeg  -f lavfi -i testsrc="size=1920x1080:rate=50" -t 10 -pix_fmt yuv420p -f null -`
jtop you need check HW engine state

cizekmilan, 2020-06-06:
yes, NVENC shows 384MHz (instead of OFF). One from CPU cores still utilized to 100% with -pix_fmt yuv420p. I will verify on real input stream.

Is it possible show de/encoder utilization on Jetson (something similar nvidia-smi)?

Good job!

Mizbauddin, 2020-07-03:
Hi @jocover while encoding from MJPEG to H264, I see that ffmpeg used 100 % of all the cpu cores and no GPU usage. Also NVENC and NVDEC are off. Is this normal for ffmpeg to use cpu for encoding or does it use gpu for encoding? Thank you.

yystylechen, 2021-01-07:
https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%2520Linux%2520Driver%2520Package%2520Development%2520Guide%2Fmultimedia.html%23wwpID0E0JB0HA
This document may be helpful！