jocover/jetson-ffmpeg #44: Need to add more params to manual

ethaniel, 2020-06-07:
`Encoder h264_nvmpi [nvmpi H.264 encoder wrapper]:
    General capabilities: delay hardware
    Threading capabilities: none
    Supported pixel formats: yuv420p
h264_nvmpi_encoder AVOptions:
  -num_capture_buffers <int>        E..V..... Number of buffers in the capture context (from 1 to 32) (default 10)
  -profile           <int>        E..V..... Set the encoding profile (from -99 to 100) (default -99)
     baseline                     E..V.....
     main                         E..V.....
     high                         E..V.....
  -level             <int>        E..V..... Profile Level (from 0 to 62) (default auto)
     auto                         E..V.....
     1.0                          E..V.....
     1.1                          E..V.....
     1.2                          E..V.....
     1.3                          E..V.....
     2.0                          E..V.....
     2.1                          E..V.....
     2.2                          E..V.....
     3.0                          E..V.....
     3.1                          E..V.....
     3.2                          E..V.....
     4.0                          E..V.....
     4.1                          E..V.....
     4.2                          E..V.....
     5.0                          E..V.....
     5.1                          E..V.....
  -rc                <int>        E..V..... Override the preset rate-control (from -1 to INT_MAX) (default -1)
     cbr                          E..V..... Constant bitrate mode
     vbr                          E..V..... Variable bitrate mode
  -preset            <int>        E..V..... Set the encoding preset (from 1 to 4) (default default)
     default                      E..V.....
     slow                         E..V.....
     medium                       E..V.....
     fast                         E..V.....
     ultrafast                    E..V.....
`

jocover, 2020-06-09:
nvidia will add encoding ffmpeg support

`Hi, The current implementation supports video decoding only. We will add video encoding in future release.`
https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%2520Linux%2520Driver%2520Package%2520Development%2520Guide%2Fmultimedia.html%23wwpID0ERHA

icebear74, 2020-08-19:
Hi,
the Info, that Encoding will be implemented later, is gone .. :(