jocover/jetson-ffmpeg #47: AFTER LAST UPDATE I'M GETTING nvmpi error. ANY CLUE? Thanks.

hugoceledon, 2020-06-16:
Following the steps it used to work for me.
now:

git clone https://github.com/jocover/jetson-ffmpeg.git
cd jetson-ffmpeg
mkdir build
cd build
cmake .. <--- I GET THIS OUTPUT ON TERMINAL HERE

-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Configuring done
CMake Error at CMakeLists.txt:17 (add_library):
  Cannot find source file:

    /usr/src/jetson_multimedia_api/samples/common/classes/NvBuffer.cpp

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx

CMake Error: CMake can not determine linker language for target: nvmpi
CMake Error: Cannot determine link language for target "nvmpi".
-- Generating done
-- Build files have been written to: /home/nano1/src/jetson-ffmpeg/build

jocover, 2020-06-17:
yes ,only support JetPack 4.3 or above
JetPack 4.2 please use old version

hugoceledon, 2020-06-17:
Thank you very much.

closed 2020-06-17 by hugoceledon