jocover/jetson-ffmpeg #48: Sometimes (1 of 20-40 runs) encoder regularly return packets with old timestamps

teplofizik, 2020-06-18:
`nvmpi_encoder_put_frame: 1683 us
nvmpi_encoder_get_packet: 1650 us
nvmpi_encoder_put_frame: 1716 us
nvmpi_encoder_get_packet: 1683 us
nvmpi_encoder_put_frame: 1749 us
nvmpi_encoder_get_packet: 1716 us
nvmpi_encoder_put_frame: 1782 us
nvmpi_encoder_get_packet: 1749 us
nvmpi_encoder_put_frame: 1815 us
nvmpi_encoder_get_packet: 1452 us <<<
[flv @ 0x55c3b1e880] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 1749 >= 1452`

Like circular buffer's error, but I don't know.

teplofizik, 2020-06-19:
Rough bugfix:
```
int nvmpi_encoder_get_packet(nvmpictx* ctx,nvPacket* packet){

	int ret,packet_index;

	if(ctx->packet_pools->empty())
		return -1;

	packet_index= ctx->packet_pools->front();
	
	auto ts = ctx->timestamp[packet_index];
	auto size = ctx->packets_size[packet_index];
	if((ts > 0) && (size == 0)) // Old packet, but 0-0 skip!
	{
		// This block will be executed 4-6 times continuosly on start of encoding process and after that will not be executed ever.
		//printf("READ: %lu, size: %u   == DROP ==\n", ts, size);

		return -1;
	} 
	// printf("READ: %lu, size: %u\n", ts, size);
	packet->payload=ctx->packets[packet_index];
	packet->pts=ts;
	
	packet->payload_size=size;
	if(ctx->packets_keyflag[packet_index])
		packet->flags|= 0x0001;//AV_PKT_FLAG_KEY 0x0001

	ctx->packets_size[packet_index] = 0; // mark as readed
	ctx->packet_pools->pop();
	
	return 0;
}
```

Log of this func:
```
READ: 33, size: 180798
READ: 33, size: 1055
READ: 66, size: 88100
READ: 99, size: 7312
READ: 132, size: 3536
READ: 165, size: 3378
READ: 198, size: 2517
READ: 231, size: 4478
READ: 264, size: 8260
READ: 297, size: 12519
READ: 330, size: 3594
READ: 363, size: 14102
READ: 396, size: 3786
READ: 429, size: 8334
READ: 462, size: 77579
READ: 495, size: 6099
READ: 528, size: 2275
READ: 231, size: 0   == DROP ==
READ: 231, size: 0   == DROP ==
READ: 231, size: 0   == DROP ==
READ: 231, size: 0   == DROP ==
READ: 231, size: 0   == DROP ==
READ: 231, size: 0   == DROP ==
READ: 561, size: 3494
READ: 594, size: 3862
READ: 594, size: 1537
READ: 594, size: 8767
READ: 594, size: 12399
```
After that encoding process will be performing without problems.

teplofizik, 2020-06-19:
Patches for 41 and 48 issues. Maybe they will be useful for someone.

[patches.zip](https://github.com/jocover/jetson-ffmpeg/files/4803889/patches.zip)

jocover, 2020-06-22:
Thank you for fixing the bug