jocover/jetson-ffmpeg #54: Missing info from README.md

danboid, 2020-09-09:
Hi @jocover !

Thanks v.much for developing jetson-ffmpeg! I was hoping to use my nano as a media playback device but it was pretty useless for that until I discovered jetson-ffmpeg! Very impressive work!

There are at least 2 things I would like to see added to the README:

The decoding example command isn't very useful. I think it should be replaced by a couple of working ffplay examples such as these:

```
ffplay -vcodec h264_nvmpi -i vid.mkv
ffplay -vcodec hevc_nvmpi -noautorotate -i vid2.mp4
```

I say 'working' but I can only get ffplay to work with nvmpi if I run it as root. Do you know what I need to do to give my regular user permission to use nvmpi? My user is already in the video group. If there is an existing solution to this, I would also like to see it added to the README.

I also have a few of other questions that are unrelated to the documentation:

Do you know if nvmpi hardware decoding will work with mpv without having to rebuild or update the L4T mpv package?

Have you created a debian (source) package for jetson-ffmpeg yet? I'm sure it'll be easy enough to tweak the focal ffmpeg source package to create one for jetson-ffmpeg.

I have read on the NVidia dev forums that the official L4T build of ffmpeg is supposed to support hardware decoding now but I haven't got it to work, have you?

Thanks!

danboid, 2020-09-10:
Here are the instructions for building ffmpeg with the official decoding patch:

http://ffmpeg.org/pipermail/ffmpeg-devel/2020-June/263746.html

I presume that will get merged into ffmpeg instead of jetson-ffmpeg, because its from NVIDIA? Shame it doesn't support encoding.

I've spoken to the mpv devs and I'm told nvmpi playback should work under mpv if it works under ffmpeg.

Should it be possible to build and run jetson-ffmpeg without the tegra libraries, say for example if anyone gets Arch, Debian or regular arm64 Ubuntu running on the jetson? That would be another advantage as it seems the official ffmpeg patch depends on the Tegra libraries.

danboid, 2020-09-12:
I think I've answered most of my own questions about jetson-ffmpeg etc now.

I have just tried to build jetson-ffmpeg on Armbian Focal (running on my nano) but running cmake `configure` fails because I don't have LIB_NVBUF installed. Do you think libnvbuf would be installable on alternate Tegra distros or do you think you could maybe re-work jsff so that it doesn't require nvbuf?

I've had a quick look at the mpv source and it seems its hwdec drivers are entirely separate from ffmpeg so hardware decoding isn't going to come for free under mpv, as you probably knew.

My only real question now is how can I use jsff or the official NV accelerated ffmpeg without having to use sudo?

eusoubrasileiro, 2022-05-09:
@danboid  I tried on this jetson nano image https://github.com/pythops/jetson-nano-image that it also doesn't work on Focal. 
I will try now going back to Bionic.

danboid, 2022-05-10:
I didn't know of this custom Nano distro until now. I'm pretty busy this week but I'll try it out soon. Thanks @eusoubrasileiro !