jocover/jetson-ffmpeg #56: Can jetson tx2 compile jetson-ffmpeg

honggesmile, 2020-09-17:
first , thanks for your work,this is great.
Now i comlile this on my jetson tx2. when i run `./configure --enable-nvmpi` 
error appears:
`ERROR: nvmpi not found using pkg-config

If you think configure made a mistake, make sure you are using the latest
version from Git.  If the latest version fails, report the problem to the
ffmpeg-user@ffmpeg.org mailing list or IRC #ffmpeg on irc.freenode.net.
Include the log file "ffbuild/config.log" produced by configure as this will help
solve the problem.
`

I tried to find nvmpi in the system,  
When I type   `pkg-config --list-all | grep nvmpi`   in the command line，
`nvidia@nvidia:~/jetsonTX2-ffmpeg/ffmpeg$ pkg-config --list-all | grep nvmpi
nvmpi                            nvmpi - nvidia multimedia api
`

I  cannot find any solutions,can you give me some advices.thanks for your help.

pzhu10, 2020-10-12:
you should config the  nvmpi.pc file to pkg-config path
like this：
PKG_CONFIG_PATH=/usr/local/lib/pkgconfig
generally speaking ，if you make install jetson-ffmpeg correctly, this will be already set

[image]

you can compile jetson-ffmpeg again,
hope useful for you, lucky~~

honggesmile, 2020-10-12:
it works,thanks very much.

closed 2020-10-12 by honggesmile