jocover/jetson-ffmpeg #61: Hardware encoding with ffmpeg

Fonta88, 2020-10-15:
Hi,

I'm trying to convert the following ffmpeg pipeline (which works, tested) to make it work on a Jetson Xavier.

python3 script.py | ffmpeg -y -vsync 0 -hwaccel cuda -hwaccel_output_format cuda -f rawvideo -pix_fmt rgb24 -s:v 1920x1080 -r 24 -i - -c:v h264_nvenc -f rtsp rtsp://localhost:8554/mystream

the python script generates frame for the ffmpeg pipeline.

What parameters I must specify to perform h264 hardware encoding using h264_nvmpi instead of h264_nvenc?
I mean, ok, --c:v h264_nvenc will become --c:v h264_nvmpi, but what about other parameters as -hwaccel and -hwaccel_output_format?

simply replacing --c:v h264_nvenc with --c:v h264_nvmpi result in the following output

Input #0, rawvideo, from 'pipe:':
  Duration: N/A, start: 0.000000, bitrate: 1194393 kb/s
    Stream #0:0: Video: rawvideo (RGB[24] / 0x18424752), rgb24, 1920x1080, 1194393 kb/s, 24 tbr, 24 tbn, 24 tbc
Device creation failed: -12.
[rawvideo @ 0x55808b6390] No device available for decoder: device type cuda needed for codec rawvideo.
Stream mapping:
  Stream #0:0 -> #0:0 (rawvideo (native) -> h264 (h264_nvmpi))
Device setup failed for decoder on input stream #0:0 : Cannot allocate memory

thanks in advance for your support

GavinDarkglider, 2021-01-27:
try 

python3 script.py | ffmpeg -y -vsync 0 -f rawvideo -pix_fmt yuv420p -s:v 1920x1080 -r 24 -i - -c:v h264_nvmpi -f rtsp rtsp://localhost:8554/mystream

I have no idea if cuda hwaccel actually works on the jetson board as it sits in ffmpeg.

Camofelix, 2021-01-27:
> [rawvideo @ 0x55808b6390] No device available for decoder: device type cuda needed for codec rawvideo.

What this is saying is that ffmpeg has no cuda device to send the decoded video frames to. There's a larger overview below, but essentially ffmpeg doesnt understand the jetson's memory structures because the jetson doesnt use cuda memory addressing the same way as every other nvidia product since cuda 1.0 

Can you try adding -log_level verbose to both commands and adding that below? either directly or in a pastebin is fine.

 Nvidia has yet to implement it in their own build AFAIK and this repo only deals with accessing the hardware encoding blocks and decoding blocks.

I'm waiting for my jetson's to arrive and will then be attempting to get hw accelerated cuda filters (and by extension cuda frame types) working on the nano/jetsons (something I need for a project that will involve this repository)

dakaix, 2021-01-29:
@Camofelix Looking forward to that if you can get hwaccel's to work, I need that functionality for Zoneminder to properly use acceleration since you cannot specify input/output codec's.