jocover/jetson-ffmpeg #62: Request for Dockerfile

stevemac00, 2020-10-25:
Thanks so much for jetson-ffmpeg and it works great.

I have attempted to create a Dockerfile and get to "Install the project..." but I don't know which libraries are needed from /usr/lib/aarch64-linux-gnu/tegra/ and don't want to copy them all.

Is there any chance you could make a Dockerfile?

KillahB33, 2021-06-08:
I am working on this now for another project, the libraries has been a pain but I'll report back with the base image I am using as I am trying to find the correct one. If you want the image small though you may need to pull out the libraries you need.

spattinson, 2021-11-02:
There is a lot of possibly unnecessary stuff needed due to dependencies installing libraries, most of what is needed is in these two packages `nvidia-l4t-jetson-multimedia-api nvidia-l4t-multimedia` but you end up with dependencies `nvidia-l4t-cuda nvidia-l4t-3d-core nvidia-l4t-wayland libffi6` 

Full Dockerfile I use with Frigate NVR - ffmpeg has a minimal set of options for my env:
```
FROM        balenalib/jetson-nano-ubuntu:bionic AS base

WORKDIR     /tmp/workdir

ENV     DEBIAN_FRONTEND=noninteractive
RUN	apt-get -yqq update && \
	mkdir -p /opt/nvidia/l4t-packages/ && \
	touch /opt/nvidia/l4t-packages/.nv-l4t-disable-boot-fw-update-in-preinstall && \
	echo '#!/bin/sh\nsu -c "$*"' > /usr/local/bin/sudo && chmod +x /usr/local/bin/sudo && \
        echo "/usr/lib/aarch64-linux-gnu/tegra" > /etc/ld.so.conf.d/aarch64-linux-gnu_GL.conf && ldconfig && \
	apt-get -y -o Dpkg::Options::=--force-confdef -o Dpkg::Options::=--force-confnew install \
        --no-install-recommends ca-certificates expat libgomp1 cmake xz-utils rsync gnupg git wget libv4l-dev libegl1-mesa-dev \
	nvidia-l4t-jetson-multimedia-api nvidia-l4t-multimedia nvidia-l4t-cuda nvidia-l4t-3d-core nvidia-l4t-wayland libffi6 && \
        apt-get autoremove -y && \
        apt-get clean -y

ENV UDEV=1

FROM base as build

ENV     FFMPEG_VERSION=4.3.1 \
        FREETYPE_VERSION=2.11.0 \
        LAME_VERSION=3.100 \
        LIBPTHREAD_STUBS_VERSION=0.4 \
        SRC=/usr/local

ARG         LD_LIBRARY_PATH=/opt/ffmpeg/lib
ARG         MAKEFLAGS="-j4"
ARG         PKG_CONFIG_PATH="/opt/ffmpeg/share/pkgconfig:/opt/ffmpeg/lib/pkgconfig:/opt/ffmpeg/lib64/pkgconfig"
ARG         PREFIX=/opt/ffmpeg
ARG         LD_LIBRARY_PATH="/opt/ffmpeg/lib:/opt/ffmpeg/lib64:/usr/lib64:/usr/lib:/lib64:/lib"

#RUN	echo "deb https://repo.download.nvidia.com/jetson/common r32.6 main" \
#	 > /etc/apt/sources.list.d/nvidia-l4t-apt-source.list && \
#	echo "deb https://repo.download.nvidia.com/jetson/t210 r32.6 main" \
#	 >> /etc/apt/sources.list.d/nvidia-l4t-apt-source.list
#
#RUN	apt-key adv --fetch-key https://repo.download.nvidia.com/jetson/jetson-ota-public.asc && \

RUN     buildDeps="autoconf \
        automake \
        curl \
        bzip2 \
        libexpat1-dev \
        g++ \
        gcc \
        gperf \
        libtool \
        make \
        nasm \
        perl \
        pkg-config \
        python \
        libssl-dev \
        yasm \
        libomxil-bellagio-dev \
        zlib1g-dev" && \
        apt-get -yqq update && \
        apt-get install -yq --no-install-recommends ${buildDeps}
#
## Nvidia Jetson hwaccel https://github.com/jocover/jetson-ffmpeg
RUN	DIR=/tmp/jetson-ffmpeg && \
	 mkdir -p ${DIR} && \
	 cd ${DIR} && \
	 git clone https://github.com/jocover/jetson-ffmpeg.git && \
	 cd jetson-ffmpeg && \
[... 68 lines cut ...]
RUN \
        ldd ${PREFIX}/bin/ffmpeg | grep opt/ffmpeg | cut -d ' ' -f 3 | xargs -i cp {} /usr/local/lib/ && \
        for lib in /usr/local/lib/*.so.*; do ln -s "${lib##*/}" "${lib%%.so.*}".so; done && \
        cp -rv /opt/ffmpeg/lib/* /usr/local/lib/ && \
        cp ${PREFIX}/bin/* /usr/local/bin/ && \
        cp -r ${PREFIX}/share/ffmpeg /usr/local/share/ && ls /usr/local/lib && \
	ldd /usr/local/bin/ffmpeg && \
#        LD_LIBRARY_PATH=/usr/local/lib ffmpeg -buildconf && \
        cp -r ${PREFIX}/include/libav* ${PREFIX}/include/libpostproc ${PREFIX}/include/libsw* /usr/local/include && \
        mkdir -p /usr/local/lib/pkgconfig && \
        for pc in ${PREFIX}/lib/pkgconfig/libav*.pc ${PREFIX}/lib/pkgconfig/libpostproc.pc ${PREFIX}/lib/pkgconfig/libsw*.pc; do \
        sed "s:${PREFIX}:/usr/local:g" <"$pc" >/usr/local/lib/pkgconfig/"${pc##*/}"; \
        done

FROM        base AS release

ENV         LD_LIBRARY_PATH=/usr/local/lib:/usr/local/lib64:/usr/lib:/usr/lib64:/lib:/lib64:/usr/lib/aarch64-linux-gnu/tegra:/usr/lib/aarch64-linux-gnu/tegra-egl

CMD         ["--help"]
ENTRYPOINT  ["ffmpeg"]

COPY --from=build /usr/local /usr/local/

# Run ffmpeg with -c:v h264_nvmpi to enable HW accell for decoding on Jetson Nano
```

closed 2022-03-31 by stevemac00