jocover/jetson-ffmpeg #64: Assertion failed when decoding+encoding h265

rhpk, 2020-12-12:
Hi, I'm trying to decode/encode with ffmpeg, here is my command line:

ffmpeg/jetson-ffmpeg/ffmpeg/ffmpeg -c:v hevc_nvmpi -i swiv.mkv -preset slow -c:v hevc_nvmpi -b:v 9000k -c:a copy swiv_enctest2.mp4

I get:

Assertion ((src_linesize) >= 0 ? (src_linesize) : (-(src_linesize))) >= bytewidth failed at libavutil/imgutils.c:314

Using software decoding, it works fine, thought CPU is 100% and I can only encode at 8.5 fps:

ffmpeg/jetson-ffmpeg/ffmpeg/ffmpeg -i swiv.mkv -preset slow -c:v hevc_nvmpi -b:v 9000k -c:a copy swiv_enctest2.mp4

rhpk, 2020-12-14:
I tried to play a bit with patched ffmpeg and here's what I get. Basically, src_linesize < bytewidth which makes the assertion fail. This is, of course ffmpeg 4.2.

image_copy_plane -------------------------------------------------------------------------------------
src_linesize = 3840
dst_linesize = 7680
bytewidth    = 7680

rhpk, 2020-12-22:
I did a bit of debugging with gdb and the issue boils down to nvmpi_decode() function. When nvmpi_decoder_get_frame() is called, the _nvframe.linesize[0] gets filled with 3840 instead of 7680. Maybe I'm doing something wrong with ffmpeg command line which is fooling the hardware decoder around.

syadnom, 2021-01-05:
-c:v hevc_nvmpi should be to the right of the source file...