jocover/jetson-ffmpeg #65: h.265 encode performance?

syadnom, 2021-01-05:
I've seen some documentation that the jetson nano is capable of 4x 1080p hevc encodes.  Has anyone been able to confirm this with this ffmpeg build?

AnterCreeper, 2021-11-19:
about 3.5x-4.1x.
input:h264 nvdec
output: hevc nvenc -preset medium -b:v 6000k -profile:v high(this one maybe useless)

AnterCreeper, 2021-11-19:
The real speed depends on the content of your input. But at least, higher than 3.0x.

gjrtimmer, 2026-06-14:
HEVC encode performance improvements available in the [gjrtimmer/jetson-ffmpeg](https://github.com/gjrtimmer/jetson-ffmpeg) fork:

- **Issue**: gjrtimmer/jetson-ffmpeg#9
- **Commit**: [`d8bd111`](https://github.com/gjrtimmer/jetson-ffmpeg/commit/d8bd111)
- **Change**: Enable `setMaxPerfMode` for encoder — lifts per-session NVENC clock governor (DFS). Was hardcoded true, now configurable via `-max_perf 0/1`. Applies to both H.264 and HEVC encoders. Up to 6x speedup on non-MAXN power profiles.
- **Validated**: smoke-all.sh 7/7 FFmpeg versions (4.2–8.0), hw-perf-mode + hw-perf-bench on Orin Nano