jocover/jetson-ffmpeg #66: Instant segmentation fault

dlueth, 2021-01-13:
`./ffmpeg.nvmpi -i rtp://some-h264-rtp-stream -vf deblock -c:v h264_nvmpi -c:a copy -f null -`

leads to

`Segmentation fault (core dumped)`

in 99% of cases instantly. Compiled on Jetson Nano exactly from README.md on JetPack 4.4.1. This is just a stripped-down example to check if it works at all.

Any idea what's wrong?

GavinDarkglider, 2021-01-25:
Try the second nvmpi patch in #69

GavinDarkglider, 2021-01-25:
That patch is a quick and dirty fix that allows apps that arent ffmpeg that allow you to define the decoder used to use the decoder, even if the player isnt setting the pix_fmt, because it is assumed it is set in the codec init.

dlueth, 2021-02-16:
@GavinDarkglider sorry for the late reply. The second patch does not work as well. Not segfaulting but as soon as I start encoding I get broken pipe.