jocover/jetson-ffmpeg #67: Output of encoder to cuda frame format

Camofelix, 2021-01-22:
Partially a feature request- ability to decode the frame to an ffmpeg cuda frame so that it can be used in conjunction with cuda HW filters directly, instead of going back and fourth between cpu and gpu context. As of now we end up doing 6 copies of memory structure's over the PCIe bus, instead of 2 if we kept everything in the GPU context with cuda frames.            

Diagram below to show current flow: 
[image]

pjueon, 2022-04-14:
@FCLC 
Hi. This might be a stupid question, but is it possible to use `scale_cuda` filter with this jetson-ffmpeg?

FCLC, 2022-04-14:
> 

I'm the wrong person to ask, I've also not worked on this in a few months time. If you have the hardware, best to test and report back so it can be documented.