jocover/jetson-ffmpeg #68: Extended color range (0-255)

prophetzopu, 2021-01-22:
Hello,

I am using jetson-ffmpeg with ffmpeg 4.2.1 on a Jetson AGX Xavier. My inputs are RTSP and RTMP streams in AV_PIX_FMT_YUVJ420P format. The decoder claims to be decoding to AV_PIX_FMT_YUV420P format with color_range AVCOL_RANGE_JPEG.

As far as I can tell, the color_range is actually AVCOL_RANGE_MPEG (limited, 16-235). If I put the image data into an OpenCV cv::Mat, then do a limited color to full color scale operation (e.g.`original.convertTo(processed,` CV_8UC1, 256.0f/220.0f, -16.0f);`) then dump the image, colors look as I'd expect.

I would assume similar issues would occur if I attempt to encode in full range.

Thoughts/suggestions?

This project has been great so far. Thank you for your time!

prophetzopu, 2021-01-25:
Here is a histogram dump of what I am seeing with software decode vs hardware decode. This doesn't seem like typical full to limited conversion because the middle region seems unaffected.
[image]
[image]

mhm0902, 2021-08-19:
How to support yuvj420 decoding？thank you！

grantthomas, 2021-08-19:
> How to support yuvj420 decoding？thank you！

https://github.com/jocover/jetson-ffmpeg/issues/83
https://github.com/jocover/jetson-ffmpeg/issues/80

When I apply the advice in issue 83, I can use yuvj420p, but transcoding to mp4 containers hangs ffmpeg.
When I apply the advice in issue 80, I can transcode to mp4 ffmpeg, but yuvj420p doen't work.

I'm sure there's a combo of commits that makes it all work.