jocover/jetson-ffmpeg #69: Most apps that can use decoder fail on ff_get_buffer due to pixfmt not being set properly.

GavinDarkglider, 2021-01-25:
After some serious testing, we, in the nintendo switch scene have found this issue. We have also noticed that Nvidia's ffmpeg codec also has the same issue. We were able to work around this poorly, by setting avctx->pix_fmt=AV_PIX_FMT_YUV420P; in decode function before it calls ff_get_buffer. Fairly sure this is an issue in upstream ffmpeg libs, but thought since you dont support NV12 anyway, you might want the patch, so you can test yourself.

[ffmpeg_nvmpi.txt](https://github.com/jocover/jetson-ffmpeg/files/5870052/ffmpeg_nvmpi.txt)

GavinDarkglider, 2021-01-25:
#46 [ffmpeg_nvmpi.txt](https://github.com/jocover/jetson-ffmpeg/files/5870217/ffmpeg_nvmpi.txt)
Better potential fix. This one checks if PIX_FMT is set on decoder init, and if it isnt, will set it to yuv420p, and if it is, will check to see if it is yuv420p, and if not will return with error. I would assume the same would work with nv12 support, if it is ever fixed....

lzcchl, 2021-01-26:
hello, is "ffmpeg_nvmpi.txt" work? and just replace "ffmpeg_nvmpi.patch"?

GavinDarkglider, 2021-01-26:
It should, we actually solved this with the upstream nvidia decoder. you still have to be able to force the decoder you want in an app.
mpv makes a nice test app, as I know it had the issue before, and doesnt now.

Also use the patch in the second comment.

GavinDarkglider, 2021-01-26:
Actually, a few typos in that patch. give me a minute.
[ffmpeg_nvmpi.txt](https://github.com/jocover/jetson-ffmpeg/files/5870603/ffmpeg_nvmpi.txt)

Camofelix, 2021-01-26:
have you been able to test with multiple different file types? 

if so, maybe make a pull request, hopefully can be merged. 

Don't have my nano's in right now to test but as a general rule, if something can deal with the file below, it can deal with anything 

https://drive.google.com/uc?export=download&id=1T7ChytQ2iZA39M4SG81JFwxMbflUi5Cc

relevant file information: 
Resolution : 3840 x 2160
File size : 724 MB
Duration : 00:01:23
Format : MP4
Codec : HEVC
Overall bit rate : 73 Mb/s
Bit depth : 10 bits
Frame rate : 60 fps

it will come out gray because of lack of tonemaping (something I'm working on for a transcode cluster project)

GavinDarkglider, 2021-01-26:
My main focus has been porting linux to the nintendo switch.... this project when we used it for moonlight was the only option for ffmpeg that was viable and usable in standard video player apps.... have you tried with the nvidia ffmpeg patch. that doesnt use jetson-ffmpeg package? you might have better luck. but I have not tried anything crazy with it. I do know it fixed issues in apps I forced to use the decoder for both hevc, and x264. I didnt try the other decoders, because i figured if these work now, those should to.

Camofelix, 2021-01-26:
Slightly OT, but the main thing I'm looking to confirm for myself is the ability to use HW cuda filtering (for scalling, de-interlacing etc.) in combination with hardware decode and encode. 

Discussing over on the [Devel] ML it seems right now that the nvidia patch doesn't have the ability to keep the frames in the GPU frame buffer, which may cause problems with extra memory copies.

c1ngular, 2021-01-26:
@GavinDarkglider  hardware acceleration will work on mpv player ?

GavinDarkglider, 2021-01-28:
yes, check my pull request, and pull the patch file. It fixes it for firefox, mpv player, vlc, etc. mpv defaults to mplayer decoder, not ffmpeg, so you have to to -vd=<x>_nvmpi for the <x> decoder you want

c1ngular, 2021-01-28:
@GavinDarkglider  thank you .

closed 2021-01-30 by GavinDarkglider