jocover/jetson-ffmpeg #72: undefined reference to `ff_vp8_nvmpi_decoder'

cnavarrete, 2021-01-28:
I have installed this patch a lot of times, but this is the first time I get this error, I'm using a Jetson Xavier NX by the way, and I have done the same procedure in the same jetson before.

 ``` 
(lots of lines here)

./libavformat/avformat.h:885:21: note: declared here
     AVCodecContext *codec;
                     ^~~~~
fftools/ffmpeg.c:3998:13: warning: ‘codec’ is deprecated [-Wdeprecated-declarations]
             input_streams[i]->st->codec->debug = debug;
             ^~~~~~~~~~~~~
In file included from fftools/ffmpeg.c:43:0:
./libavformat/avformat.h:885:21: note: declared here
     AVCodecContext *codec;
                     ^~~~~
LD	ffmpeg_g
libavcodec/libavcodec.a(allcodecs.o):(.data.rel.ro+0x1308): undefined reference to `ff_vp8_nvmpi_decoder'
collect2: error: ld returned 1 exit status
Makefile:111: recipe for target 'ffmpeg_g' failed
make: *** [ffmpeg_g] Error 1

 ```

cnavarrete, 2021-01-28:
Just realized I have the same problem as https://github.com/jocover/jetson-ffmpeg/issues/71#issue-796094626

jankais3r, 2021-01-28:
Which JetPack version are you on? Just trying to figure out if this could be linked to upgrade to 4.5 or not...

cnavarrete, 2021-01-28:
I'm still on 4.4

dakaix, 2021-01-28:
I had the same problem, managed to fix it by editing libavcodec/nvmpi_dec.c and adding the following at the bottom:

NVMPI_DEC(vp8, AV_CODEC_ID_VP8,NULL);

Make runs happily afterwards.

cnavarrete, 2021-01-29:
Thanks gonna check if it works, could you tell me what it does please?

dakaix, 2021-01-29:
It adds the definition for that codec, it seems that VP8 was missed out on the most recent version in the repo. That's why the error you get is that it cannot find the definition for the vp8 nvmpi decoder.

jocover, 2021-01-29:
Thank you for reminding and fixing the issue

dakaix, 2021-01-29:
No worries, sadly I then hit the segmentation fault issue afterwards - however the patch in [Issue 69](https://github.com/jocover/jetson-ffmpeg/issues/69) seems to be working for that now too.

cnavarrete, 2021-01-29:
It works, thanks for everything.

closed 2021-01-29 by cnavarrete

GavinDarkglider, 2021-01-30:
It appears, when I sent my fixes, I screwed that up... sorry, easily fixed, thanks for pointing it out.