jocover/jetson-ffmpeg #73: hevc_nvmpi decoder error: Invalid Pix_FMT for NVMPI

jankais3r, 2021-01-29:
I am unable to use the `hevc_nvmpi` decoder as it gives the following error. Encoding with `hevc_nvmpi` works just fine, as well as decoding using the stock `hevc` decoder.

```
user@jetson:~$ ffmpeg -codecs | grep nvmpi
 DEV.LS h264                 H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10 (decoders: h264 h264_v4l2m2m h264_nvmpi ) (encoders: h264_nvmpi libx264 libx264rgb h264_omx h264_v4l2m2m h264_vaapi )
 DEV.L. hevc                 H.265 / HEVC (High Efficiency Video Coding) (decoders: hevc hevc_nvmpi hevc_v4l2m2m ) (encoders: hevc_nvmpi libx265 hevc_v4l2m2m hevc_vaapi )
 DEV.L. mpeg2video           MPEG-2 video (decoders: mpeg2video mpegvideo mpeg2_v4l2m2m mpeg2_nvmpi ) (encoders: mpeg2video mpeg2_vaapi )
 DEV.L. mpeg4                MPEG-4 part 2 (decoders: mpeg4 mpeg4_v4l2m2m mpeg4_nvmpi ) (encoders: mpeg4 libxvid mpeg4_v4l2m2m )
 DEV.L. vp8                  On2 VP8 (decoders: vp8 vp8_v4l2m2m libvpx vp8_nvmpi ) (encoders: libvpx vp8_v4l2m2m vp8_vaapi )
 DEV.L. vp9                  Google VP9 (decoders: vp9 vp9_v4l2m2m libvpx-vp9 vp9_nvmpi ) (encoders: libvpx-vp9 vp9_vaapi )

user@jetson:~$ ffmpeg -c:v hevc_nvmpi -loglevel debug -i test.mov rgb/rgb_%06d.png
ffmpeg version c9f3835 Copyright (c) 2000-2020 the FFmpeg developers
  built with gcc 7 (Ubuntu/Linaro 7.5.0-3ubuntu1~18.04)
  configuration: --pkg-config-flags=--static --enable-rpath --extra-cflags='-I/home/user/ffmpeg_build/include -I/usr/local/cuda/include -I/usr/local/include -I/usr/local/cuda-10.2/include -I/usr/local/cuda-10.2/targets/aarch64-linux/include -I/usr/src/linux-headers-4.9.140-tegra-ubuntu18.04_aarch64/nvidia/include -I/usr/src/jetson_multimedia_api/include -I/usr/include -I/home/user/Downloads/ffmpeg_4_2 -I/home/user/Downloads/ffmpeg_4_2/include' --extra-ldflags='-L/home/user/ffmpeg_build/lib -L/usr/local/cuda/lib64 -L/usr/local/cuda-10.2/targets/aarch64-linux/lib -L/usr/local/lib -L/usr/lib -L/var/lib -L/lib -L/usr/lib/aarch64-linux-gnu/pkgconfig -L/usr/local/lib/pkgconfig' --prefix=/home/user/ffmpeg_build --enable-static --enable-gpl --enable-nonfree --enable-version3 --enable-chromaprint --enable-frei0r --enable-ladspa --enable-libaribb24 --enable-libass --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdc1394 --enable-libfdk-aac --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-liblensfun --enable-libmodplug --enable-libmp3lame --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvo-amrwbenc --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxvid --enable-libzvbi --enable-libmysofa --enable-openal --enable-opengl --disable-stripping --enable-avresample --enable-gnutls --enable-libdrm --enable-shared --enable-cuda-nvcc --enable-libnpp --enable-nvmpi --enable-libiec61883 --enable-libvidstab --enable-libaom --enable-libxml2 --enable-libbluray --enable-omx --enable-libzmq
  libavutil      56. 31.100 / 56. 31.100
  libavcodec     58. 54.100 / 58. 54.100
  libavformat    58. 29.100 / 58. 29.100
  libavdevice    58.  8.100 / 58.  8.100
  libavfilter     7. 57.100 /  7. 57.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  5.100 /  5.  5.100
  libswresample   3.  5.100 /  3.  5.100
  libpostproc    55.  5.100 / 55.  5.100
Splitting the commandline.
Reading option '-c:v' ... matched as option 'c' (codec name) with argument 'hevc_nvmpi'.
Reading option '-loglevel' ... matched as option 'loglevel' (set logging level) with argument 'debug'.
Reading option '-i' ... matched as input url with argument 'test.mov'.
Reading option 'rgb/rgb_%06d.png' ... matched as output url.
Finished splitting the commandline.
Parsing a group of options: global .
Applying option loglevel (set logging level) with argument debug.
Successfully parsed a group of options.
Parsing a group of options: input url test.mov.
Applying option c:v (codec name) with argument hevc_nvmpi.
Successfully parsed a group of options.
Opening an input file: test.mov.
[NULL @ 0x55b31d2e50] Opening 'test.mov' for reading
[file @ 0x55b316d210] Setting default whitelist 'file,crypto'
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x55b31d2e50] Format mov,mp4,m4a,3gp,3g2,mj2 probed with size=2048 and score=100
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x55b31d2e50] ISO: File Type Major Brand: qt  
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x55b31d2e50] Unknown dref type 0x73696c61 size 12
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x55b31d2e50] Processing st: 0, edit list 0 - media time: 582, duration: 38670709
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x55b31d2e50] skip 582 audio samples from curr_cts: 0
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x55b31d2e50] Unknown dref type 0x73696c61 size 12
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x55b31d2e50] Shifting DTS by 900 because of negative CTTS.
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x55b31d2e50] Processing st: 1, edit list 0 - media time: 0, duration: 5261500
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x55b31d2e50] stts: 0 ctts: 0, ctts_index: 0, ctts_count: 51693
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x55b31d2e50] stts: 100 ctts: 100, ctts_index: 1, ctts_count: 51693
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x55b31d2e50] stts: 200 ctts: -100, ctts_index: 2, ctts_count: 51693
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x55b31d2e50] stts: 300 ctts: 100, ctts_index: 3, ctts_count: 51693
…
…
...
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x55b31d2e50] stts: 5261100 ctts: 100, ctts_index: 51689, ctts_count: 51693
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x55b31d2e50] stts: 5261200 ctts: -100, ctts_index: 51690, ctts_count: 51693
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x55b31d2e50] stts: 5261300 ctts: 100, ctts_index: 51691, ctts_count: 51693
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x55b31d2e50] stts: 5261400 ctts: -100, ctts_index: 51692, ctts_count: 51693
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x55b31d2e50] Before avformat_find_stream_info() pos: 2759507978 bytes read:925102 seeks:2 nb_streams:2
[hevc @ 0x55b31da480] nal_unit_type: 32(VPS), nuh_layer_id: 0, temporal_id: 0
[hevc @ 0x55b31da480] Decoding VPS
[hevc @ 0x55b31da480] Main profile bitstream
[hevc @ 0x55b31da480] nal_unit_type: 33(SPS), nuh_layer_id: 0, temporal_id: 0
[hevc @ 0x55b31da480] Decoding SPS
[hevc @ 0x55b31da480] Main profile bitstream
[hevc @ 0x55b31da480] Decoding VUI
[hevc @ 0x55b31da480] nal_unit_type: 34(PPS), nuh_layer_id: 0, temporal_id: 0
[hevc @ 0x55b31da480] Decoding PPS
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x55b31d2e50] demuxer injecting skip 582 / discard 0
[aac @ 0x55b31d3f10] skip 582 / discard 0 samples due to side data
[aac @ 0x55b31d3f10] skip 582/1024 samples
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x55b31d2e50] All info found
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x55b31d2e50] After avformat_find_stream_info() pos: 148517 bytes read:1073583 seeks:3 frames:39
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'test.mov':
  Metadata:
    major_brand     : qt  
    minor_version   : 0
    compatible_brands: qt  
    creation_time   : 2021-01-24T09:16:31.000000Z
    com.apple.quicktime.make: Apple
    com.apple.quicktime.model: MacBookPro14,2
    com.apple.quicktime.software: macOS 11.1 (20C69)
    com.apple.quicktime.creationdate: 2021-01-24T10:01:34+0100
  Duration: 00:14:36.92, start: 0.000000, bitrate: 25174 kb/s
    Stream #0:0(und), 38, 1/44100: Audio: aac (LC) (mp4a / 0x6134706D), 44100 Hz, stereo, fltp, 245 kb/s (default)
    Metadata:
      creation_time   : 2021-01-24T09:16:31.000000Z
      handler_name    : Core Media Audio
    Stream #0:1(und), 1, 1/6000: Video: hevc (Main), 1 reference frame (hvc1 / 0x31637668), yuvj420p(pc, bt709, progressive), 2880x1800 (2880x1808) [SAR 1:1 DAR 8:5], 0/1, 24920 kb/s, 58.95 fps, 60 tbr, 6k tbn, 6k tbc (default)
    Metadata:
      creation_time   : 2021-01-24T09:16:31.000000Z
      handler_name    : Core Media Video
      encoder         : HEVC
Successfully opened the file.
Parsing a group of options: output url rgb/rgb_%06d.png.
Successfully parsed a group of options.
Opening an output file: rgb/rgb_%06d.png.
Successfully opened the file.
[hevc_nvmpi @ 0x55b32e1fe0] Invalid Pix_FMT for NVMPI
Stream mapping:
  Stream #0:1 -> #0:0 (hevc (hevc_nvmpi) -> png (native))
Error while opening decoder for input stream #0:1 : Unknown error occurred
[AVIOContext @ 0x55b3169680] Statistics: 1073583 bytes read, 3 seeks
```

I tried to manually set the Pix_FMT using `-pix_fmt yuvj420p`, but that just gives me error `Option pixel_format not found.`. I tried several HEVC videos from different sources, all result in the same error.

Anyone else seeing this?

GavinDarkglider, 2021-01-30:
Currently, the only pixel format really supported is yuv420p, hince why if an app sets it to something else, it gives that error, and bombs in decoder init. NV12 is also supported, but those havnt been pulled yet.

jankais3r, 2021-01-31:
Ah right, now I can see this in the patch. Thank you.

closed 2021-01-31 by jankais3r