jocover/jetson-ffmpeg #75: Force Decoding of Yuv420p10- Truncate to 8 bit needed

Camofelix, 2021-02-11:
For cases where 10 bit signals may be sent to the decoder, the encoder on the nano cannot output 10 bit content. it can however accept a 10 bit source, truncate the data and output to 8 bit

AnterCreeper, 2021-11-19:
It will generate strange picture.

FCLC, 2021-11-23:
Yes and no. Assuming the truncation is on the least significant bits, we lose minor detail.

The issue is in how different algorithms allocate bandwidth to different areas of the luminance curve. 

In a linear scale we'd be bit shifting everything by 2 equally and having a linear change 

But since HEVC, h264, and vp9 amongst other codecs dedicate more bandwidth to the lower parts of the curve that the upper parts, we can "afford" to lose more of the lower detail bits, as it is partially compensated by the disproportionately increased amount of data allocated to those sections. 

If the truncation assumes the wrong type of endianness in its implementation, then the picture will be awful, as the wrong bits will have been "promoted" to the wrong area

mattangus, 2021-11-28:
It seems that [l4t multimedia api supports 10 bit](https://docs.nvidia.com/jetson/l4t-multimedia/group__ee__nvbuffering__group.html#ggaae53b45fe3f04b8f9135cb80baeac6e4a4de0818f45efde4ace1c36bb54c26b98). Am I interpreting this correctly? I'm going to try and add this feature but I'm really not sure what I'm doing 😅

FCLC, 2021-12-07:
> It seems that [l4t multimedia api supports 10 bit](https://docs.nvidia.com/jetson/l4t-multimedia/group__ee__nvbuffering__group.html#ggaae53b45fe3f04b8f9135cb80baeac6e4a4de0818f45efde4ace1c36bb54c26b98). Am I interpreting this correctly? I'm going to try and add this feature but I'm really not sure what I'm doing sweat_smile
it is in the API, yes!