jocover/jetson-ffmpeg #80: FFmpeg hangs while attempting to transcode a h264 video

lordjim1985, 2021-03-22:
Hi,

can You advise why after sucessfull patching and buiding ffmpeg hangs while trying to transcode a h264 -> h264 video?

FFmpeg command line used:

./ffmpeg -i bbb_sunflower_1080p_30fps_normal.mp4 -c:v h264_nvmpi -b:v 5000k -b:a 128k -c:a aac -f mp4 test.mp4

I get the following:

Opening in BLOCKING MODE
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 4 
===== NVMEDIA: NVENC =====
NvMMLiteBlockCreate : Block : BlockType = 4 
875967048
842091865
H264: Profile = 77, Level = 51 
Got 0 size buffer in capture

And ffmpeg just sits there.

I'm using JetPack 4.5.1.

If I change the ffmpeg settings to create a hevc video it properly starts to transcode.

manswiss, 2021-03-25:
Same by me. I try now to revert the changes from 18. Mrz and if it works then.

manswiss, 2021-03-25:
Strange, it also not work when I revert changes from 18. Mrz, and also when I revert changes from 1. Mrz.
But I try the tool in the past (Feb 2021) and the it was working.

lordjim1985, 2021-03-25:
I've installed all of the updates that were available. When I'll get back home I'll try to recompile it with updated dependecies.

manswiss, 2021-03-25:
> I've installed all of the updates that were available. When I'll get back home I'll try to recompile it with updated dependecies.
I've tryed now with a blank jetson image without any updates. Same result.
Now I will compile the jetson-ffmpeg without commits from 18.Mrz. Maybe it works then.

manswiss, 2021-03-25:
I've tryed now with a blank jetson image without any updates. Same result.
Now I will compile the jetson-ffmpeg without commits from 18.Mrz. Maybe it works then.

manswiss, 2021-03-25:
It also not work. I don't know what is the source of the problem.

lordjim1985, 2021-03-25:
Maybe it's a profile and level problem. I'll check different settings when I'll get back.

lordjim1985, 2021-03-25:
I've checked different source files and profiles and levels. I'm afraid it did not help. I'll try to compile again.

lordjim1985, 2021-03-25:
I've tried recompiling but also ffmpeg hans after entering the command.
I've been thinking (maybe it's a clue), when gstreamer transcodes he states that:

`NvMMLiteOpen : Block : BlockType = 279
NVMEDIA: Reading vendor.tegra.display-size : status: 6
NvMMLiteBlockCreate : Block : BlockType = 279
Redistribute latency...
NvMMLiteOpen : Block : BlockType = 4
===== NVMEDIA: NVENC =====
NvMMLiteBlockCreate : Block : BlockType = 4
H264: Profile = 100, Level = 0`

While ffmpeg states:

`Opening in BLOCKING MODE
Opening in BLOCKING MODE
NvMMLiteOpen : Block : BlockType = 4
===== NVMEDIA: NVENC =====
NvMMLiteBlockCreate : Block : BlockType = 4
875967048
842091865
H264: Profile = 100, Level = 51
Got 0 size buffer in capture`

The Level part is interesting. Whatever profile I set, gstreamer always report Level = 0. Maybe something with the source libraries is broken or the've changed some options. From what I have seen in gstreamer also cabac-entropy-coding=1 parameter does not work. So that might be a clue that something is broken in the source.

pickledgator, 2021-04-08:
Anyone happen to find a resolution to this?

pi-null-mezon, 2021-04-27:
It could be fixed by checkout to 4ba259a

```
git clone https://github.com/jocover/jetson-ffmpeg.git && \
cd jetson-ffmpeg && \
git checkout 4ba259a && \
mkdir build && cd build && \
cmake .. && \
make && \
sudo make install && \
sudo ldconfig && \
git clone git://source.ffmpeg.org/ffmpeg.git -b release/4.2 --depth=1 && \
cd ffmpeg && \
cp ../ffmpeg_nvmpi.patch . && \
git apply ffmpeg_nvmpi.patch && \
./configure --enable-nvmpi && \
make && \
sudo make install
```

IsaiahPapa, 2021-04-28:
Having the same issues with most recent build [3dff99c](https://github.com/jocover/jetson-ffmpeg/tree/3dff99c3c2ef280209244d9bae5771997d4aa770). I also tried building on [4ba259a](https://github.com/jocover/jetson-ffmpeg/commit/4ba259a0c732cf8823506e4e422e51ca78eb177e) that @pi-null-mezon brought up with no luck. It idles on `Got 0 size buffer in capture e=       0kB time=00:00:00.14 bitrate=   2.6kbits/s speed=0.0795x`

**Full Console:**

```
> /home/nano/jetson-ffmpeg/build/ffmpeg/ffmpeg \
> -i /home/nano/Downloads/test_1080p.mp4 \
> -c:v h264_nvmpi /home/nano/Downloads/re-encoded_test_1080p.mp4 \
> -b 2500k -minrate 850k -maxrate 3000k

ffmpeg version c9f3835 Copyright (c) 2000-2020 the FFmpeg developers
  built with gcc 7 (Ubuntu/Linaro 7.5.0-3ubuntu1~18.04)
  configuration: --enable-nvmpi --enable-libpulse
  libavutil      56. 31.100 / 56. 31.100
  libavcodec     58. 54.100 / 58. 54.100
  libavformat    58. 29.100 / 58. 29.100
  libavdevice    58.  8.100 / 58.  8.100
  libavfilter     7. 57.100 /  7. 57.100
  libswscale      5.  5.100 /  5.  5.100
  libswresample   3.  5.100 /  3.  5.100
Trailing options were found on the commandline.
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '/home/nano/Downloads/test_1080p.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf56.36.100
  Duration: 00:02:30.13, start: 0.000000, bitrate: 10680 kb/s
    Stream #0:0(und): Video: h264 (High) (avc1 / 0x31637661), yuv420p, 1920x1080 [SAR 1:1 DAR 16:9], 10424 kb/s, 24 fps, 24 tbr, 12288 tbn, 48 tbc (default)
    Metadata:
      handler_name    : VideoHandler
    Stream #0:1(und): Audio: aac (LC) (mp4a / 0x6134706D), 48000 Hz, stereo, fltp, 255 kb/s (default)
    Metadata:
      handler_name    : SoundHandler
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (native) -> h264 (h264_nvmpi))
  Stream #0:1 -> #0:1 (aac (native) -> aac (native))
Press [q] to stop, [?] for help
Opening in BLOCKING MODE
Opening in BLOCKING MODE
NvMMLiteOpen : Block : BlockType = 4
===== NVMEDIA: NVENC =====
NvMMLiteBlockCreate : Block : BlockType = 4
875967048
842091865
H264: Profile = 77, Level = 51
NVMEDIA_ENC: bBlitMode is set to TRUE
Opening in BLOCKING MODE
Opening in BLOCKING MODE
NvMMLiteOpen : Block : BlockType = 4
===== NVMEDIA: NVENC =====
NvMMLiteBlockCreate : Block : BlockType = 4
875967048
842091865
H264: Profile = 77, Level = 51
Output #0, mp4, to '/home/nano/Downloads/re-encoded_test_1080p.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf58.29.100
    Stream #0:0(und): Video: h264 (h264_nvmpi) (avc1 / 0x31637661), yuv420p, 1920x1080 [SAR 1:1 DAR 16:9], q=-1--1, 2000 kb/s, 24 fps, 12288 tbn, 24 tbc (default)
    Metadata:
      handler_name    : VideoHandler
      encoder         : Lavc58.54.100 h264_nvmpi
    Stream #0:1(und): Audio: aac (LC) (mp4a / 0x6134706D), 48000 Hz, stereo, fltp, 128 kb/s (default)
    Metadata:
      handler_name    : SoundHandler
      encoder         : Lavc58.54.100 aac
Got 0 size buffer in capture e=       0kB time=00:00:00.14 bitrate=   2.6kbits/s speed=0.0795x
```

cnavarrete, 2021-04-29:
Ok I had the same problem and this is how I fixed it: You need to checkout to [7baa3d8](https://github.com/jocover/jetson-ffmpeg/tree/7baa3d8232478b82799a90c37b86a986e4df1df2), also you need to apply the patch from this version. 
```
wget https://raw.githubusercontent.com/jocover/jetson-ffmpeg/7baa3d8232478b82799a90c37b86a986e4df1df2/ffmpeg_nvmpi.patch
git apply ffmpeg_nvmpi.patch
```

jocover, 2021-04-30:
It may be caused by FFmpeg frame linesize is different with NVBuffer，
I temporarily canceled this submission

IsaiahPapa, 2021-04-30:
> It may be caused by FFmpeg frame linesize is different with NVBuffer，
> I temporarily canceled this submission

Not too savvy with this tech, but what @cnavarrete said works with regards to building and patching from [7baa3d8](https://github.com/jocover/jetson-ffmpeg/tree/7baa3d8232478b82799a90c37b86a986e4df1df2). I just did a fresh build about 10 minutes ago and it seems to be working great. 

Thanks guys!