jocover/jetson-ffmpeg #81: New nvmpi patch, that forces decoders.

GavinDarkglider, 2021-04-22:
This is not something I plan on merging, mainly because in its current implementation, it will break decoders if the patch was applied, and nvmpi was not enabled, or it wasnt for a jetson-ffmpeg supported platform.

All this patch does, is what the other one does, but when apps request decoder by name or id, and they are for the generic software decoder/encoder it forces ffmpeg to switch to return the nvmpi decoders/encoders. Patch included as txt file, github doesnt like patch files.

[ffmpeg_nvmpi.txt](https://github.com/jocover/jetson-ffmpeg/files/6361117/ffmpeg_nvmpi.txt)

hax0rsaw, 2021-05-17:
I assume the steps for patching this are the same as the other nvpmi patch? When I attempt this - I have errors....

GavinDarkglider, 2021-05-17:
I might have rebased the patch on a different version of ffmpeg for lakka/libreelec builds I am working on for the switch, It should be the same, that is 4.4 with a bunch of patches for kodi.

GavinDarkglider, 2022-02-16:
Have a better way to do this, will pr once I have all of the changes orgainzed.

closed 2022-02-16 by GavinDarkglider