jocover/jetson-ffmpeg #83: Invalid Pix_FMT for NVMPI Only yuv420p is supported

xlordz, 2021-05-04:
As I saw in other closed issues, the patched version can work with yuvj420p, but in my deploy it isn't supported.
Nano with Jetpack 4.5.1.

ffmpeg version c9f3835 Copyright (c) 2000-2020 the FFmpeg developers
  built with gcc 7 (Ubuntu/Linaro 7.5.0-3ubuntu1~18.04)
  configuration: --enable-nonfree --enable-shared --enable-nvmpi --enable-libnpp --extra-cflags=-I/usr/local/cuda/include --extra-ldflags=-L/usr/local/cuda/lib64
  libavutil      56. 31.100 / 56. 31.100
  libavcodec     58. 54.100 / 58. 54.100
  libavformat    58. 29.100 / 58. 29.100
  libavdevice    58.  8.100 / 58.  8.100
  libavfilter     7. 57.100 /  7. 57.100
  libswscale      5.  5.100 /  5.  5.100
  libswresample   3.  5.100 /  3.  5.100
Input #0, mpeg, from '00000000297000000.264':
  Duration: 00:41:54.60, start: 51709.276933, bitrate: 3395 kb/s
    Stream #0:0[0x1e0]: Video: h264 (Main), yuvj420p(pc, bt709, progressive), 1920x1080 [SAR 1:1 DAR 16:9], 15 fps, 15 tbr, 90k tbn, 30 tbc
[h264_nvmpi @ 0x556c70d750] Invalid Pix_FMT for NVMPI Only yuv420p is supported
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (h264_nvmpi) -> h264 (h264_nvmpi))
Error while opening decoder for input stream #0:0 : Invalid data found when processing input

GavinDarkglider, 2021-05-11:
Build before this commit. 
https://github.com/jocover/jetson-ffmpeg/commit/20067187641389ba309bd3ca51933718b6b475ef

Support for yuvj420p was removed there.

drakorg, 2021-07-13:
Hi, I'm facing the same problem. Picking the previous commit will hopefully solve the issue, which I am doing as I write this, but do you know why support for that pixel format has been removed?
Thank you,
Best regards.