jocover/jetson-ffmpeg #86: NVENC munmap_chunk(): invalid pointer

ghollisjr, 2021-06-04:
I've had two possibly related memory errors occur while trying to use NVENC for h264 with this ffmpeg patch on an NVIDIA Jetson Nano 2 GB.

For a ~600 MB VOB file, I get:

double free or corruption (out)

The output MP4 file has working audio but scrambled video with green diagonal lines going across what appears to be distorted diagonal versions of the source video.

For a 5 second sample of the original input VOB file, I get:

munmap_chunk(): invalid pointer

The smaller output file has the same diagonal distortion and green diagonal lines in the video output.

I'm running the same ffmpeg version on my desktop which transcodes the same videos with no errors or distortion using NVENC, so it seems the issue is not with the files but with ffmpeg running this patch on the jetson nano.

ghollisjr, 2021-06-04:
UPDATE: Looks like it's a problem with dimensions of the output video.  The input video has dimensions 720x480.  If I manually set the output dimensions to 1920x1080, then no memory errors and the video output looks fine.  If I try any other resolution, it seems ffmpeg dies with the above memory errors.

RichProgrammerEE, 2023-05-12:
Was there ever any update to this information? Is there a resolution limitation with the underlying NVIDIA multimedia API or is this a limitation with this patch/wrapper?

rjbrown99, 2023-07-04:
+1 same issue with me, double free or corruption.

I also tried with the Keylost fork with ffmpeg 6:
https://github.com/Keylost/jetson-ffmpeg/issues

That did not help, same issue. 

I also backported the latest Ubuntu x265 package (3.5+1-f0c1022b6) and recompiled ffmpeg to use it but that also didn't resolve the issue.