jocover/jetson-ffmpeg #88: motion interoprability

mithrandir42, 2021-07-12:
I have tried to get this to work for a rtsp stream with motion:

netcam_decoder hevc_nvmpi

This failes with:
[3:ml3] [INF] [NET] [Jul 12 18:04:15] netcam_rtsp_open_context: High resolution: Opened camera((null))
[2:av0] [ERR] [NET] [Jul 12 18:04:15] netcam_rtsp_decoder_error: Normal resolution: avcodec_find_decoder_by_name: Failed,Interrupt False
[2:av0] [NTC] [NET] [Jul 12 18:04:15] netcam_rtsp_decoder_error: Normal resolution: Ignoring user requested decoder hevc_nvmpi
[2:ml2] [INF] [ENC] [Jul 12 18:04:15] ffmpeg_avcodec_log: deprecated pixel format used, make sure you did set range correctly

The documentations says "Users can find the names of the decoders on their system by going to a command prompt and typing ffmpeg -decoders and looking for the name of the decoder with a 'D' next to it and correlating this with the type of image provided from the camera."

Well, the decoder does not have the D next to it, "ffmpeg -decoders | grep hevc" shows:

 VFS..D hevc                 HEVC (High Efficiency Video Coding)
 V..... hevc_nvmpi           hevc (nvmpi) (codec hevc)
 V..... hevc_v4l2m2m         V4L2 mem2mem HEVC decoder wrapper (codec hevc)
 V..... hevc_nvv4l2dec       hevc (nvv4l2dec) (codec hevc)
 
Using "netcam_decoder hevc" or no setting works, but obviously uses no hardware acceleration.

Maybe this is the reason it does not work with motion. However, calling ffmpeg manually works and uses acceleration:

ffmpeg -c:v hevc_nvmpi -i rtsp://user:Pass@192.168.XXX.XXX:554/1/h264major test.avi

I am unsure why we need (do we?) direct rendering here. Might also be unrelated. 

Also tried with plain nvida hevc_nvv4l2dec, with same result: https://github.com/Motion-Project/motion/issues/1360

Still unsure what is going wrong here. What can be done to get this to work?

eusoubrasileiro, 2021-12-19:
This is not a issue related to ffmpeg or nvidia-ffmpeg or this port even.
Read the docs of motion again. Even if a decoder is implemented by ffmpeg doesn't mean motion can you it because motion side is not implemented. You are going to do like me. Use video loopback to play that rtsp stream and video for linux from motion side.

mithrandir42, 2021-12-19:
Well, that is timing ;)

After a longer break in trying this, I just got it to work "somehow" yesterday. The clue is, we have to build motion against the self-built ffmpeg library and to uninstalll some dev and libavxxx packages in order to get it found (build and runtime). Otherwise the preinstalled packages seem to get preferred even when passing the ffmpeg path to ./configure (we have to). But I still have to figure out how to reproduce the build to give precise instructions...

This way I can get the cameras started using the custom decoders with motion config:
netcam_decoder hevc_nvmpi
or
netcam_decoder  nvida hevc_nvv4l2dec (this one segfaults somewhere in initilization)

However, hevc_nvmpi works in motion, and hw decoding is being shown in jtop, but results are not overwhelming. Motion CPU usage gets down to 250% from 300% (for 3 HD cams and  3 old vga cams, the old cams using cpu decoding, alone they need around 30%)

Have been hoping for more... Well, and I noticed, it only works when starting motion from the command line. When starting motion as a service via start-stop-daemon only the 3 old cams work shortly and then the motion process vanishes. Still have to sort this out, also...

eusoubrasileiro, 2021-12-20:
Awesome! Just checked it works perfectly!  Thanks!
`netcam_params decoder=hevc_nvmpi,rtsp_transport=tcp` works perfectly.
So I was wrong. The format this ffmpeg implements decoding is supported by motion. Awesome!