jocover/jetson-ffmpeg #94: Unknown encoder 'h264_nvmpi'

yueyihua, 2021-08-21:
desktop:~/3rd/ffmpeg$ ffmpeg -r 12 -s 640*480 -i /dev/video0 -c:v h264_nvmpi camera.mp4
ffmpeg version 3.4.8-0ubuntu0.2 Copyright (c) 2000-2020 the FFmpeg developers
  built with gcc 7 (Ubuntu/Linaro 7.5.0-3ubuntu1~18.04)
  configuration: --prefix=/usr --extra-version=0ubuntu0.2 --toolchain=hardened --libdir=/usr/lib/aarch64-linux-gnu --incdir=/usr/include/aarch64-linux-gnu --enable-gpl --disable-stripping --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librubberband --enable-librsvg --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libopencv --enable-libx264 --enable-shared
  libavutil      55. 78.100 / 55. 78.100
  libavcodec     57.107.100 / 57.107.100
  libavformat    57. 83.100 / 57. 83.100
  libavdevice    57. 10.100 / 57. 10.100
  libavfilter     6.107.100 /  6.107.100
  libavresample   3.  7.  0 /  3.  7.  0
  libswscale      4.  8.100 /  4.  8.100
  libswresample   2.  9.100 /  2.  9.100
  libpostproc    54.  7.100 / 54.  7.100
Input #0, video4linux2,v4l2, from '/dev/video0':
  Duration: N/A, start: 91206.075533, bitrate: 147456 kb/s
    Stream #0:0: Video: rawvideo (YUY2 / 0x32595559), yuyv422, 640x480, 147456 kb/s, 30 fps, 30 tbr, 1000k tbn, 1000k tbc
Unknown encoder 'h264_nvmpi'

My config command：
./configure --enable-nvmpi --enable-encoder=h264_nvmpi,hevc_nvmpi --enable-decoder=h264_nvmpi,hevc_nvmpi,mpeg2_nvmpi,mpeg4_nvmpi,vp8_nvmpi,vp9_nvmpi --prefix=./install
or
./configure --enable-nvmpi --prefix=./install

dbussert, 2021-08-22:
if you do a `ffmpeg -encoders | grep nv` you should see if `h264_nvmpi` was added.  The header info on your ffmpeg command does not maybe the configuration flags you listed. That indicates to me you're not running the ffmpeg you compiled and simply getting the default Ubuntu ffmpeg

grantthomas, 2021-08-23:
I agree with dbussert, it looks like you're running the system ffmpeg.
your command `ffmpeg .....` will run the system installed version by default, even if you're in the same directory as you compiled it in. To run the version in your directory use `./fffmpeg .....`. This tells the system explicitly to use the local directory as the source.

A couple quick tips if you're not familiar with linux.

- run this command to see what ffmpeg you're using: `which ffmpeg`
- the system will use the executable which is in the first entry in your PATH environment variable.
  - run `echo $PATH` to see this.
- to run your compiled version either:
  - `/full/path/to/ffmpeg ...`
  - `./ffmpeg` if in current directory with local compile
  - place your compiled version somewhere in your PATH

yueyihua, 2021-08-24:
sorry，I have not install the ffmpeg I have build. It is resolved.

closed 2021-08-24 by yueyihua