jocover/jetson-ffmpeg #95: H264 Encode 1920x1080 just 5fps

yueyihua, 2021-08-25:
Input #0, video4linux2,v4l2, from '/dev/video0':
  Duration: N/A, start: 435852.495003, bitrate: 199065 kb/s
    Stream #0:0: Video: rawvideo (YUY2 / 0x32595559), yuyv422, 1920x1080, 199065 kb/s, 6 fps, 6 tbr, 1000k tbn, 1000k tbc
Stream mapping:
  Stream #0:0 -> #0:0 (rawvideo (native) -> h264 (h264_nvmpi))
Press [q] to stop, [?] for help
Opening in BLOCKING MODE
NvMMLiteOpen : Block : BlockType = 4
===== NVMEDIA: NVENC =====
NvMMLiteBlockCreate : Block : BlockType = 4
875967048
842091865
H264: Profile = 77, Level = 51
NVMEDIA_ENC: bBlitMode is set to TRUE
Opening in BLOCKING MODE
NvMMLiteOpen : Block : BlockType = 4
===== NVMEDIA: NVENC =====
NvMMLiteBlockCreate : Block : BlockType = 4
875967048
842091865
H264: Profile = 77, Level = 51
Output #0, mp4, to 'camera1080.mp4':
  Metadata:
    encoder         : Lavf58.29.100
    Stream #0:0: Video: h264 (h264_nvmpi) (avc1 / 0x31637661), yuv420p, 1920x1080, q=-1--1, 2000 kb/s, 30 fps, 15360 tbn, 30 tbc
    Metadata:
      encoder         : Lavc58.54.100 h264_nvmpi
NVMEDIA_ENC: bBlitMode is set to TRUE
frame=  414 fps=5.0 q=-0.0 Lsize=    3150kB time=00:00:13.76 bitrate=1874.3kbits/s dup=4 drop=0 speed=0.167x

grantthomas, 2021-08-25:
Can you post your full command?

I've snipped some of your reply.

> Input #0, video4linux2,v4l2, from '/dev/video0':
> Stream #0:0: Video: rawvideo (YUY2 / 0x32595559), yuyv422, 1920x1080, 199065 kb/s, 6 fps, 6 tbr, 1000k tbn, 1000k tbc
> encoder : Lavf58.29.100
> Stream #0:0: Video: h264 (h264_nvmpi) (avc1 / 0x31637661), yuv420p, 1920x1080, q=-1--1, 2000 kb/s, 30 fps, 15360 tbn, 30 
> frame= 414 fps=5.0 q=-0.0 Lsize= 3150kB time=00:00:13.76 bitrate=1874.3kbits/s dup=4 drop=0 speed=0.167x

It looks like your source is 6fps, If so, 5 is not as bad as opposed to 30 fps assumed in your output stream.
You can try adding `-r <fps> ` before your `-i /dev/video0` and see what happens.

yueyihua, 2021-08-30:
sorry，my source is just 6fps, this only can be test through USB3.0.

closed 2021-08-30 by yueyihua

Porter-X, 2022-09-27:
It's just because the default resolution radio 1920x1080 is too high, our arm board can not provide enough resources to support UVC. So we need to use the FFmpeg command to set the resolution lower like " ffmpeg -f v4l2  -r 25 -s 320x240 -i /dev/video0  -s 320x240 -r 25 -f flv rtmp://127.0.0.1/live/wayne ",hope this comment could help you!