jocover/jetson-ffmpeg #106: nvmpi not enabled in ffmpeg when using --enable-shared

G4GUO, 2021-12-10:
I am trying to use hardware decoding with https://github.com/robagar/h264decoder
It seems as is if this library requires avcodec to be built as a shared object.
If I try to build ffmpeg with --enable-nvmpi --enable-shared, the version of ffmpeg I get does not 
have the nvmpi codecs included, and I get a message telling me there is a clash between the two options.
This only seems to apply to avcodec.lib.so all the other ones build fine as shared.
If I build it as static there is no problem but it is not what I need.
jetson-ffmpeg builds both static and shared libraries so it should be possible.
I have got a bit stuck on this, so suggestions would be welcomed

eusoubrasileiro, 2021-12-19:
I just did it. You need to export to the `LD_LIBRARY_PATH` where you installed the library.

For example, if you installed using `./configure  ... --prefix=/usr/local`

```
LD_LIBRARY_PATH=/usr/local/lib
export LD_LIBRARY_PATH
```

G4GUO, 2021-12-19:
> I just did it. You need to export to the `LD_LIBRARY_PATH` where you installed the library.
> 
> For example, if you installed using `./configure ... --prefix=/usr/local`
> 
> ```
> LD_LIBRARY_PATH=/usr/local/lib
> export LD_LIBRARY_PATH
> ```

Do you need to set the library path when you are building  https://github.com/robagar/h264decoder ?
or when you are building ffmpeg? I get the incompatibility warning when I build ffmpeg as shared.

eusoubrasileiro, 2021-12-19:
@G4GUO  I am not sure what you are trying to do? Or if you are confused about where to open issues. 
This is a issue for a ffmpeg project targeting jetson nano nvidia. From the link you mention that's something for raspberry pi. 
I am not getting what you are trying to do...

G4GUO, 2021-12-20:
I am trying to build libavcodec.so with nvmpi support.

I didn't want to open an issue here, but as this is not an NVIDIA supported project where else could I ask for help?

The issue seems to be the combination of using  --enable-nvmpi --enable-shared when building the patched version of ffmpeg.

It builds ok but when I try to use it I get a message saying --enable-nvmpi --enable-shared are incompatible options and when I list the codecs the nvidia hardware ones are missing.

(The linked project is simply a Python callable h.264 decoder based on libavcodec, it works fine on my nano/XavierNX hardware
but of course doesn't support NVIDIA hw decoding at present)

G4GUO, 2021-12-20:
It does build a new shared version but when I run the ffmpeg it creates I get told the option of --enable-nvmpi and --enable-shared are inconsistent. All the required input .so files exist. I suspect something in the make/cmake process is not right. I was hoping someone else had seen this and knew of a quick fix. Thanks for taking the time to look at it.

eusoubrasileiro, 2021-12-20:
If you read calmly what I wrote and research about it you will find the answer to your problem. As I said your are not exporting the built libraries. Linux doesn't know where they are so its using the one they know. Unfortunately I don't have time to walk you through.

G4GUO, 2021-12-20:
To the best of my knowledge I am exporting the built libraries correctly, on my first attempt I wasn't, but I corrected that weeks ago. Let's simply agree to disagree. I need to re-exam the patch file, as while it works perfectly with a static build, it does not seem to work correctly with a shared build of ffmpeg.