jocover/jetson-ffmpeg #109: git apply ffmpeg_nvmpi.patch error

fhkong, 2022-03-03:
When I use the command: git apply ffmpeg_nvmpi.patch in my jetson tx2, the following error occurs!
```
error: patch failed: configure:340
error: configure: patch does not apply
error: patch failed: libavcodec/Makefile:354
error: libavcodec/Makefile: patch does not apply
error: patch failed: libavcodec/allcodecs.c:143
error: libavcodec/allcodecs.c: patch does not apply
error: libavcodec/nvmpi_dec.c: already exists in working directory
error: libavcodec/nvmpi_enc.c: already exists in working directory
```

anonymity1, 2022-06-11:
I have the same question when I used the command: git apply ffmpeg_nvmpi.patch in my jetson nano :(

gaojulong, 2022-06-13:
Jetson Nano :

error: patch failed: configure:340
error: configure: patch does not apply
error: patch failed: libavcodec/Makefile:354
error: libavcodec/Makefile: patch does not apply
error: patch failed: libavcodec/allcodecs.c:143
error: libavcodec/allcodecs.c: patch does not apply
error: libavcodec/nvmpi_dec.c: already exists in working directory
error: libavcodec/nvmpi_enc.c: already exists in working directory

eusoubrasileiro, 2023-04-13:
You are applying the patch twice! If you check the patch and the files you will see it was applied already.