jocover/jetson-ffmpeg #112: New Decoder/encoder available with pretty much all issues fixed, and huge speed increases.

GavinDarkglider, 2022-03-23:
Main tree:
https://gitlab.com/switchroot/switch-l4t-multimedia/FFmpeg

FFMPEG replacement deb for bionic, built against bionic ffmpeg sources to keep app compatibilty.
https://gitlab.com/l4t-community/gnu-linux/switchroot-pipeline/-/jobs/2211932828/artifacts/download?file_type=archive

VLC replacement deb(Needed a patch from upstream): https://gitlab.com/l4t-community/gnu-linux/switchroot-pipeline/-/jobs/2211932829/artifacts/download?file_type=archive

Replacement libffmpeg for chromium, sorry, no package yet: https://gitlab.azka.li/l4t-community/gnu-linux/switchroot-pipeline/-/jobs/647/artifacts/download?file_type=archive

For chromium, you should edit .desktop files, replace ```%U``` with  ```--no-sandbox --use-vulkan --enable-features=Vulkan %U```, then find/replace libffmpeg.so with one in archive.

Dont know off the top of my head where this is stored.

Also, Chrome doesnt do vp9/av1 via ffmpeg, so you need to install a plugin called h264ify if you want decoder to work in youtube, also this disables 4k. Sorry, Blame chrome developers for this.

kevmo314, 2022-04-26:
I'm curious what's the source of the speed increase? I see that the new lib is using the V4L2 API instead of the multimedia API, is that it?

eusoubrasileiro, 2022-05-09:
@GavinDarkglider broken links there. I was looking for your faster ffmpeg. Where is that?

GavinDarkglider, 2022-05-10:
> @GavinDarkglider broken links there. I was looking for your faster ffmpeg. Where is that?

Yeah, I dont know what happened with the links, it was all gitlab ci stuff, the first link is the ffmpeg 4.4 branch, but you will need to back port it to the version from bionic if you want it to be system-wide. I will see what I can do about the links, but no promises.

Azkali, 2022-05-13:
We have a repo setup for jetson: https://jetson.repo.azka.li

xsacha, 2022-10-05:
@Azkali is it possible to publish a new build on the repo with latest changes (especially the close stream memory leak)? Thanks.