jocover/jetson-ffmpeg #113: nvmpi decoder fails with some RTSP streams

w3sip, 2022-04-12:
Trying to use nvmpi with ffmpeg 4.2 -- using commit `3dff99c3c2ef280209244d9bae5771997d4aa770` from this repository.

It seems to be very finicky with the bitstream it attempts to consume. For example:

`ffmpeg/bin/ffmpeg -c:v h264_nvmpi -i rtsp://127.0.0.1/short.mkv  -f null -` - this command fails to decode RTSP stream served by live555 (frames are never generated)

The stream is fine (a different decoder is able to process it):

`ffmpeg/bin/ffmpeg -c:v h264 -i rtsp://127.0.0.1/short.mkv  -f null -` - this command decodes it without issue

The problem isn't exclusive to live555 (we've seen it fail with ffmpeg-served video, albeit not as often), but this is the easiest way to reproduce the problem.

I'm attaching
[live555.zip](https://github.com/jocover/jetson-ffmpeg/files/8468537/live555.zip)

Just running `./live555 &` and one of the commands above should reproduce the issue.

aimaicai, 2022-04-14:
I've the same problem: if I connect to a rtsp stream from a live camera is all ok, but if I connect to a rtsp media server streaming a video in loop it doesn't work: no frame are decoded. I'm using ffmpeg 4.2 and rtsp-simple-server (https://github.com/aler9/rtsp-simple-server) as server.
If I launch ffmpeg with the debug loglevel I see this output repeatedly coming till I ctrl-c:

> cur_dts is invalid st:0 (0) [init:0 i_done:0 finish:0] (this is harmless if it occurs once at the start per stream) 

Thanks for the awesome project.

w3sip, 2022-04-14:
The problem seems to be not with this wrapper, but the actual nvidia implementation. I'm seeing the same behavior if running with nvidia-supplied ffmpeg on Jetson. Tracking it here - https://forums.developer.nvidia.com/t/nvmpi-decoder-doesnt-work/210993/7

grantthomas, 2022-04-14:
Can you elaborate on 'nvidia-supplied ffmpeg on Jetson' ?
From what I've understood, the entire reason why jetson-ffmpeg exists is because nvidia _doesn't_ provide or support ffmpeg with hardware acceleration.

w3sip, 2022-04-14:
@grantthomas - see https://forums.developer.nvidia.com/t/the-nvidia-ffmpeg-package-supports-hardware-accelerated-decode-on-jetson-platforms/123694?page=2

grantthomas, 2022-04-14:
Interesting, thank you for the link @w3sip