jocover/jetson-ffmpeg #115: Unable to compile ffmpeg in new l4t 34.1 ubuntu 20.04

aimaicai, 2022-04-19:
Hi, I've recently upgrade my jetson agx to the latest version (ubuntu 20.04 with l4t 34.1). I follow the guide to enable the nvmpi codec for ffmpeg (4.2.2), but it fail on configure:
ERROR: nvmpi not found using pkg-config
The package is present, so i check the configure.log in the ffbuild folder and I get this:
/usr/bin/ld: /usr/local/lib/libnvmpi.so: undefined reference to `NvBufSurfaceMap'
/usr/bin/ld: /usr/local/lib/libnvmpi.so: undefined reference to `NvBufSurfaceFromFd'
/usr/bin/ld: /usr/local/lib/libnvmpi.so: undefined reference to `NvBufSurfaceUnMap'
Have anyone been able to compile ffmpeg with the nvmpi codecs enabled?
Thanks

eusoubrasileiro, 2022-05-09:
@russ-go I am looking forward for your progress, on the same page as you.

vladirai, 2022-07-17:
Same issue on jetson xavier nx with the latest version (Jetpack 5.0.1, ubuntu 20.04 with l4t 34.1.1). Attaching config.log
[config.log](https://github.com/jocover/jetson-ffmpeg/files/9127211/config.log)
.

xsacha, 2022-07-19:
Has anyone done this conversion yet? So I don't redo all their work.

21ramil, 2022-09-20:
@aimaicai  I'm wondering how you solved this problem

xsacha, 2022-09-20:
Apparently nvidia has added hardware jetson decoder support to ffmpeg, 2 years ago, already.
https://forums.developer.nvidia.com/t/hardware-accelerated-video-playback-with-l4t-ffmpeg/154019/7

There's also this fork:
https://github.com/LinusCDE/mad-jetson-ffmpeg

HTani, 2023-01-17:
You can build it by adding "-L/usr/lib/aarch 64 -linux-gnu/tegra-lnvbufsurface" to Libs in jetson-ffmeg/nvmpi.pc.in.

> --- nvmpi.pc.in.org	2023-01-13 10:40:43.210327891 +0900
> +++ nvmpi.pc.in	2023-01-16 13:16:17.057770034 +0900
> @@ -8,5 +8,5 @@
>  Version: @PROJECT_VERSION@
>  
>  Requires:
> -Libs: -L${libdir} -lnvmpi
> +Libs: -L${libdir} -lnvmpi -L/usr/lib/aarch64-linux-gnu/tegra -lnvbufsurface
>  Cflags: -I${includedir} 

It works on ffmpeg-4.2.7, Ubuntu 20.04.04 LTS/Orin.

shunguang, 2023-01-18:
os: ubuntu 20.04 LTS, 
jetpack: 5.0.2-b231
ffmpeg git info:
remote.origin.url=git://source.ffmpeg.org/ffmpeg.git
remote.origin.fetch=+refs/heads/release/4.2:refs/remotes/origin/release/4.2
branch.release/4.2.remote=origin
branch.release/4.2.merge=refs/heads/release/4.2

I follow HTani's suggestion, but has new errors

----- from ./ffbuild/config.log -----------
BEGIN /tmp/ffconf.lUMlGBb1/test.c
    1   #include <nvmpi.h>
    2   #include <stdint.h>
    3   long check_nvmpi_create_decoder(void) { return (long) nvmpi_create_decoder; }
    4   int main(void) { int ret = 0;
    5    ret |= ((intptr_t)check_nvmpi_create_decoder) & 0xFFFF;
    6   return ret; }
END /tmp/ffconf.lUMlGBb1/test.c
gcc -D_ISOC99_SOURCE -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_POSIX_C_SOURCE=200112 -D_XOPEN_SOURCE=600 -DPIC -I/usr/local/cuda-11.4/include -I/usr/local/include -std=c11 -fomit-frame-pointer -fPIC -pthread -I/usr/local/include -c -o /tmp/ffconf.lUMlGBb1/test.o /tmp/ffconf.lUMlGBb1/test.c
gcc -L/usr/local/cuda-11.4/lib64 -L/usr/local/lib -L/usr/lib/aarch64-linux-gnu -Wl,--as-needed -Wl,-z,noexecstack -I/usr/local/include -o /tmp/ffconf.lUMlGBb1/test /tmp/ffconf.lUMlGBb1/test.o
/usr/bin/ld: /tmp/ffconf.lUMlGBb1/test.o: in function `check_nvmpi_create_decoder':
test.c:(.text+0x0): **undefined reference to `nvmpi_create_decoder'**
/usr/bin/ld: test.c:(.text+0x4): **undefined reference to `nvmpi_create_decoder'**
collect2: error: ld returned 1 exit status
ERROR: nvmpi not found using pkg-config
``
Any suggestions? Thanks

shunguang, 2023-01-18:
P.S.:
(1) By the following check, nvmpi_create_decoder is in the lib. 
nvidia@ubuntu:/usr/local/lib$ nm /usr/local/lib/libnvmpi.a | grep -w T | grep -E 'nvmpi_create_decoder'
0000000000000d3c T nvmpi_create_decoder
nvidia@ubuntu:/usr/local/lib$ nm /usr/local/lib/libnvmpi.so | grep -w T | grep -E 'nvmpi_create_decoder'
00000000000118e0 T nvmpi_create_decoder

(2) here is my config cmd:
/media/nvidia/pkg/ffmpeg$ ./configure --disable-static --enable-shared --enable-nonfree --enable-libnpp --enable-nvmpi --enable-gpl --disable-opencl --extra-cflags="-I/usr/local/cuda-11.4/include -I/usr/local/include" --extra-ldflags="-L/usr/local/cuda-11.4/lib64 -L/usr/local/lib -L/usr/lib/aarch64-linux-gnu"

minghaohsu410168, 2023-01-31:
> You can build it by adding "-L/usr/lib/aarch 64 -linux-gnu/tegra-lnvbufsurface" to Libs in jetson-ffmeg/nvmpi.pc.in.
> 
> > --- nvmpi.pc.in.org	2023-01-13 10:40:43.210327891 +0900
> > +++ nvmpi.pc.in	2023-01-16 13:16:17.057770034 +0900
> > @@ -8,5 +8,5 @@
> > Version: @PROJECT_VERSION@
> > Requires:
> > -Libs: -L${libdir} -lnvmpi
> > +Libs: -L${libdir} -lnvmpi -L/usr/lib/aarch64-linux-gnu/tegra -lnvbufsurface
> > Cflags: -I${includedir}
> 
> It works on ffmpeg-4.2.7, Ubuntu 20.04.04 LTS/Orin (JetPack 5.0.2).

@HTani  It works after I modified nvmpi.pc file. You save my life!! Thank you.

HTani, 2023-01-31:
@shunguang 
Have you added the path of the nvmpi.pc file to the environment variable PKG_CONFIG_PATH?

shunguang, 2023-02-03:
@HTani Thanks for your kind reply. it was a silly  mistake when  interpreted your folloowing   message.
```
-Libs: -L${libdir} -lnvmpi
+Libs: -L${libdir} -lnvmpi -L/usr/lib/aarch64-linux-gnu/tegra -lnvbufsurface

```
at here  "-Libs" and "+Libs" came from your git diff,  notthe contents in  your nvmpi.pc.in file. When i changed them as
```
# Libs: -L${libdir} -lnvmpi
Libs: -L${libdir} -lnvmpi -L/usr/lib/aarch64-linux-gnu/tegra -lnvbufsurface

```
in my nvmpi.pc.in, it passed. but still have some minor issues. Anyway, here is my build note
[](https://github.com/shunguang/HowTo/blob/master/how-to-build-ffmpeg-4-jetson.txt)