jocover/jetson-ffmpeg #119: Error: Can't initialize nvrm channel Error:

eusoubrasileiro, 2022-05-13:
Error: Can't initialize nvrm channel Error: 
Can't initialize nvrm channel 
Couldn't create ddkvic Session: 
Cannot allocate memory nvbuf_utils: 
Could not create Default NvBufferSession

eusoubrasileiro, 2022-05-13:
So it was that I was missing to install ... 

[Deepstream](https://developer.nvidia.com/deepstream-getting-started) and its dependency [JetPack](https://developer.nvidia.com/embedded/jetpack). 

I finally made it after using a Bionic custom image for my jetson nano 2G. [Here ](https://github.com/eusoubrasileiro/jetson-nano-image )I leave its reference for posterity. Specially: 
 - [ This custom script](https://github.com/eusoubrasileiro/jetson-nano-image/blob/bionic_master/create-rootfs-image.sh) that installs deepstream*.deb package using `dpkg -i`. 
 - It's also necessary to unpack the [jetson_multimedia_api samples](https://github.com/eusoubrasileiro/jetson-nano-image/blob/bionic_master/jetson_multimedia_api.tar.xz) to `/usr/src`

eusoubrasileiro, 2023-04-13:
@ByerRA well it seams that now I am facing the same issue after installing the same package...

ByerRA, 2023-04-13:
> @ByerRA well it seams that now I am facing the same issue after installing the same package...

To prevent "Could not create Default NvBufferSession" errors add the user(s) running FFMPEG to the "video" group then logout and back in.

sudo usermod -aG video <username>

This solved the issue for me.

eusoubrasileiro, 2023-04-14:
@ByerRA hey just figured out almost in sync with you thanks anyway!

closed 2023-04-14 by eusoubrasileiro