jocover/jetson-ffmpeg #128: use h264_nvmpi to decode a video, the export video is very strange

chenm0520, 2023-03-17:
before transfer, the video

https://user-images.githubusercontent.com/8750736/225891151-29fd9aab-5a3a-41ee-8145-0ab7ea75889a.mp4

Video parameters: time:9s, resolution ratio:406*154,Video Encoding:H.264(Main Profile)，Bit rate:273kbbs,fps 23,

after transfer, the video 

https://user-images.githubusercontent.com/8750736/225891798-823b7084-d5de-4a07-8c65-b27768506c8d.mp4

Video parameters: time:8s, resolution ratio:406*154,Video Encoding:H.264(Main Profile)，Bit rate:582kbbs,fps 23

can anyone help me sovle the issue? git me the patch is ok,or tell me when this situation will happen,so that I can avoid it by using the decoder with cpu

bmegli, 2023-03-17:
My wild guess would be:
- horizontal resolution has to be aligned to something (likely 64 or 32 bytes)
  - maybe hardware expects input like that
    - and jetson-ffmpeg implementation doesn't take it into account
  - in such cases you typically end up with memory layout where
    - horizontal width (so called linesize or stride) is different from horizontal resolution (padded to fit)

But notably, I haven't worked with Jetson for long so this is just a guess.

---------------------------------------------

If you supply more information (what Jetson, what L4T version, what jetpack version ) it may be easier for somebody to help you.

I don't think jetson-ffmpeg works correctly with every L4T/Jetpack version.

Especially after Nvidia deprecated `nvbuf_utils` API  (in favor of `NvUtils`)

bmegli, 2023-03-17:
To put it more simply:
- check if it works correctly for video width divisible by 64
  - e.g. 640x480, 1280x720, 1920x1080
- if it does check with video width non divisible by 64

chenm0520, 2023-03-19:
> To put it more simply:
> 
> * check if it works correctly for video width divisible by 64
>   
>   * e.g. 640x480, 1280x720, 1920x1080
> * if it does check with video width non divisible by 64

yes,your guess is right just add the param `-vf "pad=ceil(iw/64)*64:ceil(ih/2)*2"` is ok