jocover/jetson-ffmpeg #131: encode2 (old) FFmpeg API affecting encoder latency

bmegli, 2023-03-20:
# Short Description

FFmpeg introduced "new" API around 2016
- https://ffmpeg.org/pipermail/ffmpeg-devel/2016-March/190525.html
- which decoupled encoder input/output

The old API (`encode2`) generally can return only single packet per-frame (unless draining).

Cold encoder init may take a lot of time (100 ms order).

By the time we get first packet from encoder:
- a few packets are are already queued and ready
- but only one of them is returned on each encode
- and the latency that sets initially is kept until end of encoding session

bmegli, 2023-03-20:
Above is only important for low-latency streaming (real-time)

bmegli, 2023-03-20:
Functional fix is in #132 
- fixes latency issues

It has other drawback - may miss last few packets in case of EOS.

Choose your poison.

Keylost, 2023-03-23:
No need to choose poison) I have added some measures to prevent packet loss